robotology/robots-configuration #1: Syncing configuration files with SW/FW

pattacini, 2016-09-01:
Following up https://github.com/robotology/icub-main/issues/360, and in particular this [comment](https://github.com/robotology/icub-main/issues/360#issuecomment-240828872) by @lornat75, before proceeding with the use of this new repo, we need a **run-time mechanism** to check consistency among `FW`, `yarprobotinterface` and **configuration files**.

It should suffice to **implement/verify** that when e.g. one critical new parameter is missing from within the conf file, then `yarprobotinterface` will abort launching the robot, warning the user to do an update.

cc @randaz81 @marcoaccame @lornat75 @vtikha @drdanz @traversaro

randaz81, 2016-09-21:
Each device is responsible for checking its own parameters: i.e. yarprobotinterface is not in charge to check robot configuration; embObjMotionControl (from icub-main) verifies boards configuration pararms, AnalogWrapper (from yarp) verifies its own params etc.
It thus the programmer which has the responsibility of implementing a correct check on the relevant params of each device, using yError() if the param is mandatory, yWarning() if the param is not present but a default fallback value is provided etc 
Generally speaking, this policy is the one already adopted in our devices, so this should not prevent us from using this new repo.

pattacini, 2016-09-21:
Cool 👍 
We can move on then!

I'll do everything tomorrow:
- Remove robots from `icub-main`
- Advertise the use of the new repo

closed 2016-09-21 by pattacini