robotology/robots-configuration #2: CMake default values to compile software

pattacini, 2016-09-01:
It would be nice to have a file containing the **robot-dependent default values** with which to **configure cmake on the `PC104`**.

cc @lorejam @drdanz @barbalberto @randaz81

pattacini, 2016-09-22:
cc @mbrunettini

drdanz, 2017-01-17:
All you need to do is to add a `.cmake` file for each repository that you need to configure containing the options that you need to enable, see for example [this file](https://github.com/robotology/yarp/blob/master/scripts/admin/initial-cache.cmake)

Then instead of just running `ccmake ..`, the first time you configure the repo you need to run `cmake -C <path>.cmake ..`

Assuming that:
* we keep these files in in robots-configuration in a subfolder for each robot with a name like `cmake_cache_<repo>.cmake
* the $CODE environment variable is set
* the $YARP_ROBOT_NAME environment variable is set
* the repository is cloned in a folder with the same name as the original repository
* the build is in a subdirectory of the git repository
* I wrote this correctly, since I didn't test it

This alias *should* do the trick:

```bash
alias cmake_first_run="cmake -C \${CODE}/robots-configuration/\${YARP_ROBOT_NAME}/cmake_cache_\$(basename \$(git rev-parse --show-toplevel)).cmake .."
```
then you can just go in your build folder and run `cmake_first_run`

pattacini, 2017-01-17:
Nice!

traversaro, 2019-11-22:
I think we can close this issue. As highlighted in https://github.com/robotology/robotology-superbuild/issues/88 , except for the cfw2can driver, with a single set of configuration flags for the projects, currently contained in the iCub Head profile (see https://github.com/robotology/robotology-superbuild#icub-head for a general  description and https://github.com/robotology/robotology-superbuild/blob/master/cmake/BuildYARP.cmake and https://github.com/robotology/robotology-superbuild/blob/master/cmake/BuildICUB.cmake for the specific options enabled) it is possible to compile all the software necessary on the icub-head . In some cases, these flags compile some devices that are not strictly necessary on that machine (such as CAN drivers on eth-only iCubs), but the gains in not compiling some code is probably not worth the maintenance burden of maintaining robot-specific configuration options.

closed 2019-11-22 by pattacini