robotology/robots-configuration #3: Missing files in iCubNancy01

serena-ivaldi, 2016-09-29:
Dear all

we just updated the robot software/firmware on the pc104, and implemented the changes suggested here: https://github.com/robotology/QA/issues/143

We installed the new robot configuration files, but we found issues and we cannot proceed.

1) when we launch robotinterface on the pc104, the robotinterface looks for robotInterface.ini but in our folder we have instead yarprobotinterface.ini.
We renamed the file to proceed temporarily - let us know if you plan to change it also in the robotinterface.

2) even fixing the file, the robotinterface crashed with a fatal error. It is looking for some configuration files that it can't find and it stops. The file generating the fatal error is: torso-ems5-mec.xml that we don't have in our hardware folder. We saw that other robots have it (Genova04, Darmstadt..) and from the logs it seems it is a new file. Could you add it to our robot too?

Thanks in advance

 @paraschos  @misaki43 @dgoepp

traversaro, 2016-09-29:
You should launch `yarprobotinterface` in place of the deprecated `robotInterface`, see https://github.com/robotology/QA/issues/109 .

For robot-specific support, it is better to use the `icub-support` repository at https://github.com/robotology/icub-support . I am going to move the issue.

traversaro, 2016-09-29:
Issue moved to [robotology/icub-support #301](https://github.com/robotology/icub-support/issues/301) via [**ZenHub**](https://www.zenhub.com/)

closed 2016-09-29 by traversaro