robotology/robots-configuration #5: Bad configuration files for the iCubNancy01 robot

misaki43, 2016-10-05:
Hello @pattacini and others,
I just add the robot configuration files, but when I launch the robot, I can see they are wrong. 
When I compare them with other robots configuration files, I can see that it is sometimes the robots name that is not define, but other times there are other parameters that are not the same (for example,  the canDeviceNum in hardware/skin/torso[...] or the kinematicTypeConnect in the cartesian folder).
Can you correct all our robot-configuration files please? Here are the logs:

```
`3 0.030356 <WARNING> Invalid syntax while loading /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./cartesian/left_arm-cartesian.xml at line 1 . "devices" element should contain the "robot" attribute 
 
4 0.030380 <WARNING> Invalid syntax while loading /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./cartesian/left_arm-cartesian.xml at line 1 . Trying to import a file for the wrong robot. Found  instead of iCubNancy01 
 
5 0.030401 <WARNING> Invalid syntax while loading /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./cartesian/left_arm-cartesian.xml at line 1 . "devices" element should contain the "build" attribute [unsigned int]. Assuming 0 
 
6 0.030420 <WARNING> Invalid syntax while loading /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./cartesian/left_arm-cartesian.xml at line 1 . Import a file for a different robot build. Found 0 instead of 1 
 
7 0.030432 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./cartesian/right_arm-cartesian.xml 
 
8 0.040355 <WARNING> Invalid syntax while loading /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./cartesian/right_arm-cartesian.xml at line 1 . "devices" element should contain the "robot" attribute 
 
9 0.040401 <WARNING> Invalid syntax while loading /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./cartesian/right_arm-cartesian.xml at line 1 . Trying to import a file for the wrong robot. Found  instead of iCubNancy01 
 
10 0.040422 <WARNING> Invalid syntax while loading /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./cartesian/right_arm-cartesian.xml at line 1 . "devices" element should contain the "build" attribute [unsigned int]. Assuming 0 
 
11 0.040435 <WARNING> Invalid syntax while loading /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./cartesian/right_arm-cartesian.xml at line 1 . Import a file for a different robot build. Found 0 instead of 1 
 
12 0.040453 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./hardware/motorControl/head-cfw2_can0-mc.xml 
 
13 0.040471 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./general.xml 
 
14 0.040489 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./hardware/electronics/head-cfw2_can0-eln.xml 
 
15 0.050425 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./hardware/mechanicals/head-cfw2_can0-mec.xml 
 
16 0.050456 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./wrappers/motorControl/head-mc_wrapper.xml 
 
17 0.060519 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./hardware/motorControl/torso-ems5-mc.xml 
 
18 0.080669 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./general.xml 
 
19 0.080713 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./hardware/electronics/torso-ems5-eln.xml 
 
20 0.080724 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./hardware/electronics/pc104.xml 
 
21 0.080734 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./hardware/mechanicals/torso-ems5-mec.xml 
 
22 0.090741 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./hardware/motorControl/torso-ems5-mc-service.xml 
 
23 0.100829 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./wrappers/motorControl/torso-mc_wrapper.xml 
 
24 0.100867 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./hardware/motorControl/left_upper_arm-ems1-mc.xml 
 
25 0.110914 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./general.xml 
 
26 0.110941 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./hardware/electronics/left_upper_arm-ems1-eln.xml 
 
27 0.110952 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./hardware/electronics/pc104.xml 
 
28 0.110961 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./hardware/mechanicals/left_upper_arm-ems1-mec.xml 
 
29 0.110970 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./hardware/motorControl/left_upper_arm-ems1-mc-service.xml 
 
30 0.120973 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./hardware/motorControl/left_lower_arm-ems2-mc.xml 
 
31 0.121003 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./general.xml 
 
32 0.121015 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./hardware/electronics/left_lower_arm-ems2-eln.xml 
 
33 0.121028 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./hardware/electronics/pc104.xml 
 
34 0.121036 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./hardware/mechanicals/left_lower_arm-ems2-mec.xml 
 
[... 236 lines cut ...]
153 0.463458 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./general.xml 
 
154 0.463506 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./calibrators/torso-calib.xml 
 
155 0.463519 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./general.xml 
 
156 0.463530 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./calibrators/right_leg-calib.xml 
 
157 0.463539 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./general.xml 
 
158 0.473524 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./calibrators/left_leg-calib.xml 
 
159 0.473550 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./general.xml 
 
160 0.473561 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./calibrators/left_arm-calib.xml 
 
161 0.473571 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./general.xml 
 
162 0.473580 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./calibrators/right_arm-calib.xml 
 
163 0.483614 <DEBUG> Reading file /usr/local/src/robot/icub-contrib-common/install-pc104/share/ICUBcontrib/robots/iCubNancy01/./general.xml `
```

Thanks!
@serena-ivaldi @olivierrochel-inria

pattacini, 2016-10-05:
@davidetome could you please deal with this request?

@misaki43 depending on the workload of @davidetome the fix might take some time. If you've already identified a bunch of errors you could think of pointing them out by opening a PR with the corresponding fixes. This will speed up the operations.

davidetome, 2016-10-06:
I've just modified a couple of files removing warnings messages about syntax errors and committed them.
It should be tested on the robot.
p.s. CanDeviceNum in hardware/skin/torso[..] is correctly settled to 9 (the CAN port number on the cfw2 board)

olivierrochel-inria, 2016-10-06:
thanks @davidetome; 
There's still a suspicious robot="iCubGenova04" in iCubNancy01/hardware/skin/torso-cfw2_can9-skin.xml, and possibly in iCub_Calibration_iCubVillersLesNancy01.xls but I can't open it. Can you check ?

davidetome, 2016-10-06:
Hi @olivierrochel-inria ,

 I removed the incorrect entry and restored/renamed the xls file that was corrupted.
Please check and let us know.

pattacini, 2016-10-13:
@misaki43 any feedback?

misaki43, 2016-10-13:
Hello @davidetome @pattacini,
sorry for this lack of feedback, now all files seems good, thanks a lot!

closed 2016-10-13 by misaki43