robotology/robots-configuration #14: iCubLondon01 needs config file update for Head V2

Tobias-Fischer, 2016-12-01:
Hi,
As discussed in https://github.com/robotology/QA/issues/167 the configuration files of `iCubLondon01` need updating, as they do not consider yet that our robot has the V2 head.
I am not sure who's best to assign this to. Any ideas @pattacini?

Thanks,
Tobi

pattacini, 2016-12-01:
@davidetome I believe the calibrators need updating, and you should also put inside `iCubLondon01` the `iKinGazeCtrl.ini` file in compliance with its new policy (see an [example](https://github.com/robotology/robots-configuration/blob/master/iCubGenova01/iKinGazeCtrl.ini)).

pattacini, 2016-12-01:
cc @randaz81

pattacini, 2016-12-01:
@Tobias-Fischer remember than to launch:

```sh
$ iKinGazeCtrl --from iKinGazeCtrl.ini
```

davidetome, 2016-12-02:
The head calibrator is already for head V2 since the robot left our labs in January 2015.
I pushed the new iKinGazeCtrl.ini file in the repo.

@randaz81

Tobias-Fischer, 2016-12-02:
Hi @davidetome,
Thanks for adding the `iKinGazeCtrl.ini`. It seems like not all config files are correct though, as `iCubGenova01` has a `HeadV2_Calibrator` (https://github.com/robotology/robots-configuration/blob/master/iCubGenova01/calibrators/head_calib.xml#L10), whereas `iCubLondon01` has a `Head_Calibrator` (https://github.com/robotology/robots-configuration/blob/master/iCubLondon01/calibrators/head_calib.xml#L9). I am not sure which other files need changes.

Thanks,
Tobi

davidetome, 2016-12-02:
@Tobias-Fischer   I think it is just the name of the device. The calibrator used is "parametricCalibrator" as well as in genova01

Tobias-Fischer, 2016-12-02:
Okay cool. I'll how about the `CALIB_ORDER`? Does that make a difference?

davidetome, 2016-12-02:
in your calibrator joints 0,1,2,3 are calibrated at the same time, then joint 4 and finally joint 5.
If you do not have any particular need leave like that.

pattacini, 2016-12-04:
Probably, we could close this.

Tobias-Fischer, 2016-12-04:
Yep, thanks!

closed 2016-12-04 by Tobias-Fischer