robotology/robots-configuration #18: Head v3 and v2.5 configuration files have the wrong names

traversaro, 2017-02-02:
Since https://github.com/robotology/robots-configuration/commit/f6bba4f288232536bae5fce4c701ccf5362903ea , the axis names in the configuration files for the Head v2.5 and v3 are wrong, i.e. do not follow the official joint names for the iCub [1]. In particular they follow the scheme `neck-pitch` (**wrong**) instead of `neck_pitch` (**correct**). 

Example of affected configurations: 
* https://github.com/robotology/robots-configuration/search?utf8=%E2%9C%93&q=neck-pitch 
* https://github.com/robotology/robots-configuration/search?utf8=%E2%9C%93&q=neck-yaw&type=Code

[1] http://wiki.icub.org/wiki/ICub_joints

cc @valegagge @davidetome @randaz81

traversaro, 2017-02-02:
This can be moved to https://github.com/robotology/icub-support , depending on the outcome of https://github.com/robotology/robots-configuration/issues/17 .

valegagge, 2017-02-02:
head v3 and head v2.5 configuration files have been copied by my experimental setup where there is the typo error that you are writing about.
Nobody noticed it and corrected it.

traversaro, 2017-02-02:
I fixed the names on iCubGenova02 do have a working system, but the problem still persists on other robots.

traversaro, 2017-02-02:
The axes names for the face were never official defined as they are not in http://wiki.icub.org/wiki/ICub_joints , but for consistency I suggest that we use underscore also in them.

davidetome, 2017-02-02:
@traversaro  I take in charge to fix other robots and close the issue when done

cc @valegagge  @randaz81

traversaro, 2017-02-02:
Thanks @davidetome !

davidetome, 2017-02-02:
fixed joints name

davidetome, 2017-02-02:
i did a mistake naming eyes_vergen and eyes_vers. i'm going to fix..again

closed 2017-02-02 by davidetome

traversaro, 2017-03-03:
cc @nunoguedelha