robotology/robots-configuration #26: master and devel not aligned

randaz81, 2017-04-10:
master and devel branches are dangerously not aligned.
@davidetome Please always merge master in devel every time you push a new commit in master.

randaz81, 2017-04-10:
Instructions here: http://www.yarp.it/md__8github_CONTRIBUTING.html
Check the paragraph _"This is the typical workflow to fix a bug in the master branch."_

traversaro, 2017-04-20:
@randaz81 @robotology/icub-developers @robotology/icub-support-team 
Originally, we did not create a `master`/`devel` branch on this repo (see https://github.com/robotology/icub-main/issues/360). We probably need to agree (in person) on how to deal with this and what is the policy for IIT robots.

pattacini, 2017-04-20:
I definitely agree to discuss this in person.

traversaro, 2017-04-20:
Just to give an example of the different workflows now in place @valegagge recently updated configuration files w.r.t. a change in the firmware only for `iCubGenova02` and `iCubGenova04` : https://github.com/robotology/robots-configuration/commit/ed89b3535a0c9b6e4cc5fb273afb13ee29596119 , as they are the only one currently on `devel` branch.

traversaro, 2017-04-20:
Another thing to discuss: should we tag this repo for releases, i.e. adding it to https://github.com/robotology/icub-main/blob/devel/admin/scripts/README_RELEASE#L54 ?

DanielePucci, 2017-05-03:
F2F discussions are indeed necessary.

closed 2018-08-01 by pattacini