robotology/robots-configuration #27: Updating robot configuration for iCubGrenoble01

EliseiFrederic, 2017-04-11:
Hi,

We are about to update our cluster to debian 8 and replace our icubsrv
by a new laptop (debian 8 or ubuntu ?)

Before, we need to push/modify the old configuration files to the new repo/syntax/structure etc.
Can we get help / remote session to perform this task ?

Thanks in advance,
Frédéric & the Grenoble team

julijenv, 2017-04-21:
Hi @EliseiFrederic ,
From your configuration files from your repository I can see that you already have the new syntax/structure to launch the robot.
The only thing that could not be applied to the repository iCubGrenoble01 on the new location on which the whole set of robot configuration files have been moved (robots-configurations) is your own local modifications, if you made some - if you have some to update just make a zip- rar file of your local repository, add it to this issue reply and we will merge it to the new location.
Let us know,
cheers

EliseiFrederic, 2017-04-21:
Hi Julien, 

Thanks for your answer. Which directories should we send to you ? 

We indeed have various modifications :
* changing the rest position, min/max for face articulators, lower gain
for one of the neck motors,
delays and ordering in the jaw/lips calibration sequence... 

I send you a zip from
/usr/local/src/robot/icub-main/build/share/iCub/robots/iCubGrenoble01 

Are there some possible modifications elsewhere ? I think we have some
camera calibrations (with lot of image data)
in 

icub@icubsrv:~/.local/share/yarp/contexts$ ls
cameraCalibration iKinGazeCtrl simCartesianControl 

icub@icubsrv:~/.local/share/yarp/contexts/cameraCalibration$ ls
calibImg_00001 calibImg_00007 calibImg_00013 icubEyes.ini
calibImg_00002 calibImg_00008 calibImg_00014 icubEyes.ini~
calibImg_00003 calibImg_00009 calibImg_00015 icubSimEyes.ini
calibImg_00004 calibImg_00010 calibImg_00016 _old_icubEyes.ini
calibImg_00005 calibImg_00011 data outputCalib.ini
calibImg_00006 calibImg_00012 icubEyes_640x480.ini

By the way, I started the reinstallation of our future icubsrv.
Following these instructions : 

http://wiki.icub.org/wiki/Linux:Installation_from_sources 

and 

http://wiki.icub.org/wiki/ICub_server_laptop_installation_instructions#Software_repositories

I see no mention about adding the robots-configuration repo. Nor
defining the YARP_ROBOT_NAME anywhere.
Also, the link to the .bashrc_iCub is bad in : 

http://wiki.icub.org/wiki/Generic_iCub_machine_installation_instructions#mount_remote_NFS_shares

Cheers,
Frederic 

> Hi @EliseiFrederic [1] ,
> From your configuration files from your repository I can see that you already have the new syntax/structure to launch the robot.
> The only thing that could not be applied to the repository iCubGrenoble01 on the new location on which the whole set of robot configuration files have been moved (robots-configurations) is your own local modifications, if you made some - if you have some to update just make a zip- rar file of your local repository, add it to this issue reply and we will merge it to the new location.
> Let us know,
> cheers 

-- 
Frederic ELISEI http://www.gipsa-lab.fr/~frederic.elisei/ 
(33-4) 76574539 mailto:/Frederic.Elisei@gipsa-lab.grenoble-inp.fr

I see no mention about adding the robots-configuration repo. Nor
defining the YARP_ROBOT_NAME anywhere.
Nor exporting any filesystem subpart to pc104 :-( 

Am I following the right instructions for a new icubsrv 

Le 21-04-2017 14:55, julijenv a écrit : 

> Hi @EliseiFrederic [1] ,
> From your configuration files from your repository I can see that you already have the new syntax/structure to launch the robot.
> The only thing that could not be applied to the repository iCubGrenoble01 on the new location on which the whole set of robot configuration files have been moved (robots-configurations) is your own local modifications, if you made some - if you have some to update just make a zip- rar file of your local repository, add it to this issue reply and we will merge it to the new location.
> Let us know,
> cheers 

-- 
Frederic ELISEI http://www.gipsa-lab.fr/~frederic.elisei/ 
(33-4) 76574539 mailto:/Frederic.Elisei@gipsa-lab.grenoble-inp.fr

 

Links:
------
[1] https://github.com/EliseiFrederic
[2]
https://github.com/robotology/robots-configuration/issues/27#issuecomment-296182965
[3]
https://github.com/

EliseiFrederic, 2017-12-13:
Hi, it looks like
 wrappers/motorControl/face_mc_wrapper.xml
is lacking a starting /      
in <param name="name">   icub/face   </param>

randaz81, 2017-12-14:
fixed, thank you.

EliseiFrederic, 2017-12-15:
Thanks @randaz81 ! I think we now have the last format for the config files...
I'll confirm the parameter order in icub_face.xml
<param name="AxisName">     "jaw"      "eyelids"    "lip-left"      "lip-up"    
 "lip-right"       "lip-bottom"        </param>  
when the robot can run again !

EliseiFrederic, 2018-01-16:
Here is how I modified 
/usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubGrenoble01/hardware/motorControl/icub_head.xml
so that yarpmotorgui displays the right labels for the talking head : 
<param name="AxisName">     "jaw"      "eyelids"    "lip-right"      "lip-up"   "lip-left"       "lip-bottom"        </param>  

Can you update the repository with this ? Thanks

closed 2018-01-16 by EliseiFrederic