robotology/robots-configuration #36: Some robots configuration files try to run parametricCalibratorETH on CAN devices, in particular on the head

traversaro, 2017-07-01:
@traversaro commented on [Fri Nov 25 2016](https://github.com/robotology/icub-main/issues/388)

We need to either: 
 * 1 : Remove that line from the configuration files, to avoid misunderstanding 
 * 2 : Add support for calibrationDelta parameter in parametricCalibrator . 

I guess @randaz81 is the one taking the decision. 

Related: https://github.com/robotology/icub-main/issues/240 , because  the documentation should describe the fine calibration procedure correctly regardless of the decision that we take. : ) 

@nunoguedelha @gabrielenava @S-Dafarra  

---

@randaz81 commented on [Fri Nov 25 2016](https://github.com/robotology/icub-main/issues/388#issuecomment-262943024)

I need few days to think about the feasibility of option 2

---

@nunoguedelha commented on [Fri Nov 25 2016](https://github.com/robotology/icub-main/issues/388#issuecomment-262959642)

@randaz81 , while waiting for your analysis, I would like to highlight the fact that the configuration of joint encoders offsets should be independent from the bus type CAN or ETHERNET. That would probably be the case also for other parameters. Maybe this would be a good opportunity to split common config parameters from those specific to the hardware interface type.

---

@randaz81 commented on [Sat Nov 26 2016](https://github.com/robotology/icub-main/issues/388#issuecomment-262989926)

There are several points wrong in this discussion, and also in the title. 
There exist two different calibrator modules, parametricCalibrator and parametricCalibratorEth, that should be used respectively on CAN and ETH devices. Each of them use different sets of parameters: obviously ETH devices are newer and they support more features, controlled by different parameters. Introducing these features on legacy devices with different MCUs may be difficult, of even impossible, without revolutionizing completely the system. So, I'm sorry, but I cannot put nothing in common between the two systems.
Additionally, here something very weird happened:
You requested to execute parametricCalibratorEth, on a CAN device (see line 5 of head-calib.xml). The system works just by luck, no surprise that these parameters have no effect on a CAN device. And since some of these parameters are mandatory for parametricCalibratorEth, you cannot even remove them from the configuration file. In the end, machines are not responsible for human negligence...

---

@traversaro commented on [Thu Dec 01 2016](https://github.com/robotology/icub-main/issues/388#issuecomment-262998192)

Title updated. 
The issue seems to affect (among the others, I did not check all the possible robots affected, just an handful that I know use ETH motor control boards except for the head): 
**Complete list of robots to fix move in last comments.**

Note that extremely recent robots (such as `iCubGenova04`) are not affected by this, as they use ETH boards even in the neck. 

---

@traversaro commented on [Sat Nov 26 2016](https://github.com/robotology/icub-main/issues/388#issuecomment-262998656)

A list of robots that could be affected is https://github.com/robotology/robots-configuration/search?utf8=%E2%9C%93&q=%3Cdevice+name%3D%22head_calibrator%22+type%3D%22parametricCalibratorEth%22%3E . The one not affected are the one that use ETH controlboards also in the neck (for sure `iCubGenova04` and `iCubLousanne02`. I am not sure, but I think that all others robots in that list are affected by this problem. 

---

@traversaro commented on [Thu Dec 01 2016](https://github.com/robotology/icub-main/issues/388#issuecomment-264151366)

Complete list of robots that need a fix:  
* [ ] `iCubGenova02` https://github.com/robotology/robots-configuration/blob/master/iCubGenova02/calibrators/head-calib.xml 
 * [ ] `iCubNancy01` https://github.com/robotology/robots-configuration/blob/master/iCubNancy01/calibrators/head-calib.xml 
 * [ ] `iCubTwente01` https://github.com/robotology/robots-configuration/blob/master/iCubTwente01/calibrators/head-calib.xml
 * [ ] `iCubDarmstad01` https://github.com/robotology/robots-configuration/blob/master/iCubDarmstadt01/calibrators/head-calib.xml
* [ ] `iCubGenova06` https://github.com/robotology/robots-configuration/blob/master/iCubGenova06/calibrators/head-calib.xml
* [ ] `iCubSheffield01` https://github.com/robotology/robots-configuration/blob/master/iCubSheffield01/calibrators/head-calib.xml
* [ ] `iCubNottingham01` https://github.com/robotology/robots-configuration/blob/master/iCubNottingham01/calibrators/head-calib.xml

cc @randaz81 @davidetome

gabrielenava, 2017-08-16:
concerning iCubNancy01, this problem should be fixed in this PR: https://github.com/robotology/robots-configuration/pull/34/commits/6ea22ece5d6e59f7eabe0a4c6bb4b28677c7d921

traversaro, 2017-08-16:
From this line https://github.com/robotology/robots-configuration/blob/f586480195638ca7e0e23452dfd1adf1aa9c2668/iCubNancy01/calibrators/head-calib.xml#L3 it seems that it is still trying to use the `parametricCalibratorEth` device for a Can device, so I don't think it was solved at all.

traversaro, 2017-08-16:
cc @valegagge that modified several of this files in the recent past.

valegagge, 2017-08-22:
The changes I done are regarding device name only.

As I already wrote here (https://github.com/robotology/icub-support/issues/500),  the problem is due to the wrong device of calibration written in the configuration files.

traversaro, 2019-10-03:
This is probably something that it could be handled by team fix @marcoaccame .

pattacini, 2023-07-30:
We just need to address the following robots:
- iCubGenova02
- iCubGenova06
- iCubNancy01

davidetome, 2023-08-10:
Done w/ this PR : 

- https://github.com/robotology/robots-configuration/pull/553

`iCubGenova02` become a full ETH robot in the meanwhile

cc @pattacini @traversaro

pattacini, 2023-08-10:
PR merged 🚀

closed 2023-08-10 by pattacini