robotology/robots-configuration #55: Why can't we directly modify the parameters in the source folder?

S-Dafarra, 2017-12-14:
It is of common use (and written in the [README](https://github.com/robotology/robots-configuration/blob/master/README.md)) to install and import in the `.local` folder the configuration files available in this repo. Nevertheless, as a result there is always discrepancy between the files in the source and those in the `.local` folder. This causes problems every time a new firmware is released and one of us (usually @nunoguedelha) has to perform by hand what git is supposed to do, i.e. version control on the `.local` files, guessing who did what and why ("*who changed the resolution of the DragonFly cameras???*") copy (again by hand) the modifications on the source folder, open a pull request and have it merged before performing a firmware update. I think it is quite a long pipeline and very error prone.
So the question is, why can't we modify and commit directly the source folder? Every robot has already its own folder, and git can help understanding who changed what and when. What is the drawback for such a solution?
@DanielePucci @traversaro

pattacini, 2017-12-14:
This is a quite important pipeline, so please let's avoid discussing on GitHub (most of the time unproductive) and plan for f2f meeting to understand your concerns @S-Dafarra

traversaro, 2017-12-14:
One problem (not saying that this should prevent us to start adopt the solution that you proposed, that I like but I need a bit more time to evaluate) is that if you modify the code in the source directory, you need to remember to run `cmake` before actually using the files.

S-Dafarra, 2017-12-14:
> plan for f2f meeting to understand your concerns

Fine for me! :slightly_smiling_face: 

> you need to remember to run cmake before actually using the files

Right! I admit I did not think about that at first, but I guess that we can add the source to the list of scanned directories, since nothing has to be compiled.

S-Dafarra, 2017-12-14:
A possible solution to make ``yarp`` look into the source folder may be to clone the ``robots-configuration`` repo in a folder named ``robots``, then the ``YARP_ROBOT_NAME`` should do the rest, once we add the containing folder to the ``YARP_DATA_DIRS``. Am I right?

pattacini, 2017-12-14:
@S-Dafarra I honestly don't like suck kind of hacking; it's very borderline ;)

pattacini, 2017-12-14:
The proposal we discussed together, i.e. no use of `.local` for robot configurations and use of git within the source plus cmake, is viable and sounds more reasonable.

S-Dafarra, 2017-12-14:
> I honestly don't like suck kind of hacking; it's very borderline ;)

I see, it is kind of stretched :grin:, but it was solving the problem of running `cmake` without any modifications. That solution was meant to provide the same "user experience" of the `.local` folder while using the `source` directory, plus it is a choice of the robot maintainer whether to allow this solution or not. @pattacini do you think it may introduce complications?

barbalberto, 2017-12-14:
My two cents
I guess the idea of having a local copy was to allow people to make their changes (temporary or not) and avoid conflicts when pulling new updates from the repo. The conflict were much more intense when configuration files were still in icub-main repo, so when user wanted to updated icub-main _code_ he/she may fall info configuration file conflict. Now they are separated, so I think conflicts are less likely to happen.

Basically you are trading the _duplicate file_ problem with the _chance_ to get occasional git conflict.
So, as long as you and the people working on the robot knows the setup and can cope with it, I think it is fine to use the source.

pattacini, 2017-12-14:
@S-Dafarra any user is supposed to be aware of the use of `.local` and `build`/`install` directories. Instead, the policy you're putting forward is unheard-of and will certainly create complications when a developer not belonging to your circle will sit down at the Greeny's console.

You might circumvent this by providing a checklist at the Greeny's desk, but it's definitely not ideal.

S-Dafarra, 2017-12-14:
@barbalberto Yes, I asked this only because every robot has its own directory. I agree that for shared repos this would not be possible. 

> Basically you are trading the duplicate file problem with the chance to get occasional git conflict.

Yes, it is the point. I rather prefer git to handle modifications rather than keeping several versions of the same file which will inevitably diverge (at a pace proportional to the number of people working on it). 

> @S-Dafarra any user is supposed to be aware of the use of .local and build/install directories. Instead, the policy you're putting forward is unheard-of and will certainly create complications when a developer not belonging to your circle will sit down at the Greeny's console.

Right, I see that it may cause confusion to people not aware of that choice. So, for the time being I will be happy of not using the `.local` folder and see whether new problems arise. 
In any case if someone needs to do some rapid tests and wants to avoid the annoyance of running `cmake`, he may still run the ``yarprobotinterface`` from the source folder, I imagine.

pattacini, 2017-12-14:
>In any case if someone needs to do some rapid tests and wants to avoid the annoyance of running cmake, he may still run the `yarprobotinterface` from the source folder, I imagine.

Correct, it's already a common practice 👍 although I don't find it particularly convincing 😏 but it's only my personal taste.

lornat75, 2017-12-14:
The current workflow (importing and modifying files in .local folder, commit permanent changes to the repository) is the result of a long discussion which took in consideration many issues. A hopefully exhaustive summary is [here](http://www.yarp.it/yarp_data_dirs.html).

For quick tests it is possible (as you say above) to run a module with full path specification or in a specific folder.

I suggest that you commit the changes you have in your .local folder in the repository so that you no longer need to modify these files. If different people use different configuration files, it is an indication that they should keep these files in their own .local folder (which, BTW, could be handled as a git repository).

S-Dafarra, 2017-12-15:
Thanks @lornat75! I am not questioning the benefits of using the ``.local`` in general. I totally agree that this workflow works pretty well with many applications. :+1: 

What I am asking is its applicability to this particular repo. In principle the ``.local`` folder belongs completely to the user and different users have different `.local`, but in this case only one user (per robot) is cloning the repository, namely the robot itself. Different physical users always access the same ``.local`` folder generated again the head of the robot. In addition I don't see the purpose of adding this ``.local`` to a repository, since it would be a duplicate of the corresponding robot folder of this repository.

lornat75, 2017-12-15:
maybe the current workflow (the same user shared by all developers on the robot) is to be revised?

pattacini, 2017-12-15:
@lornat75 as a quick summary, the possibility we discussed with @S-Dafarra and @traversaro for dealing with Greeny's setup was:

- Don't use the `.local` customization for the robots files but stick to `build`/`install` approach.
- If a modification needs to be applied to a robot file, they judge more convenient to edit the ini file in the source and then run `cmake` or `cmake && make install` to copy the file in the `build` or `build+install` directories, respectively.

This way, there is **no disruption of the current workflow** (i.e. they simply employ the `build`/`install` approach), and they can **speed up the process of keeping track of changes** (using `git` in the source) plus reporting those changes back into the source directory.

Standard-wise, we make use of `yarp-config diff` to bring a multi-folder diff of the modifications among `buid`/`install` and `.local` (on linux using meld).

My taste is that the standard procedure is still more effective, but the user can choose the other way too without any need to modify the current policy.

pattacini, 2017-12-15:
Accessibility of multiple users to multiple `.local` was not part of our previous discussion; it has been brought up perhaps by a misunderstanding.

I believe that having multiple users (different from `icub`) to deal with the robot setup will soon turn into a big mess 😉

lornat75, 2017-12-15:
then, this workflow is perfectly ok, it is in fact what you should do when you have changes that you want to commit to the repository.

valegagge, 2017-12-15:
Yes, but often user forget to update repository...maybe users should be encouraged to do so.

pattacini, 2017-12-15:
@valegagge yeah, educating users is a neverending duty 😄

valegagge, 2017-12-15:
Reading the readme file of robots-configuration repository, I think it is not enought clear that user should use local file for experiments, but it can and should change robots-configuration file for permanent changes.
More over it is important to emphasize to the user that some paremeters can be changed (e.g. pid values) while other should me modified only by manteiner (e.g. mechanical stuff).

S-Dafarra, 2017-12-15:
I don't think it is just matter of education though. Usually the use of graphical tools within the robot head is forbidden, which make difficult to compare the local files with the repository ones. The modifications than are copied and pasted by hand, and certainly not often enough. With the present workflow we end up in spending one full day every 2-3 months just to upload the modifications. I personally made a mistake once, forgetting a minus sign for a PID gain.

What I don't understand is the necessity of using the local configuration here, since it introduces mistakes, delays, slows down the development of new firmwares, with the only advantage of preserving a sort of habit.

pattacini, 2017-12-15:
@S-Dafarra go ahead with the specific flow we discussed, no problem.

Why is X forbidden on `icub-head`?
**X shall be enabled on `icub-head`** and tunneled to the console via ssh. If it's not enabled yet, please take all the necessary steps to get it work 😉

pattacini, 2017-12-15:
Btw @S-Dafarra I was not referring to you in my comment above 👍 
Nonetheless, it's a matter of fact that there are students/users who are not following the procedure just because of sloppiness, and in turn we are in charge to enforce training.

S-Dafarra, 2017-12-15:
> Why is X forbidden on icub-head?

https://github.com/robotology/QA/issues/184 

> Nonetheless, it's a matter of fact that there are students/users who are not following the procedure just because of sloppiness, and in turn we are in charge to enforce training. 

I was among them :grin:. Nevertheless I think that in case of mistakes, git is very powerful to recover :wink:.

pattacini, 2017-12-15:
@S-Dafarra ok then, there were some reasons behind, but - apparently - that post also provided a way to allow the user to get around the limitation. Hence, in the end, **X is available**.

S-Dafarra, 2017-12-15:
Nope, because the ``src`` and the ``local`` folder are in two different mount points.

valegagge, 2017-12-15:
I would give you a clarification about the use of grafical tools on icub-head.
On icub-head you could use X shell, but you shouldn't install graphical application , like gedit. 
If you need to edit something on the head you could use nano or vim or you can "export by ssh" the file on laptop and edit them using the application you prefer.
see [https://github.com/robotology/QA/issues/184](url) for more details.

The only grafical tool installed on head is meld because it used by the tool of yarp "yarp-config robot --diff".

In order to avoid copying errors , you can copy local file to repositiry file with the trivial cp command.

However, I agree with the use of the repository instead of local files to speed up our work.

valegagge, 2017-12-15:
@S-Dafarra you can mount both

pattacini, 2017-12-15:
Thank you @valegagge I clarified f2f with @S-Dafarra 
We were mixing two different threads. All clear now on both sides.

However, the discussion quickly escalated, while on GitHub we shouldn't intend posting as chatting.
Next time, for critical arguments, please consider organizing [meetings](https://github.com/robotology/robots-configuration/issues/55#issuecomment-351645112) instead 😉

valegagge, 2017-12-15:
I think that there are tools that allow us to work well enough with the robot, but they are little known or don't well documented so the user needs to make oneself comfortable.

Moreover the user feedback is usefull to improve these tools so I'm available to discuss about this.

pattacini, 2017-12-18:
Closing; feel free to further discuss this matter if you have other comments.

closed 2017-12-18 by pattacini