robotology/robots-configuration #76: Wrong hip_pitch to hip motor coupling in coupling matrices matrixM2J and matrixM2J

nunoguedelha, 2018-08-14:
After a F2F discussion with @fiorisi about the low level torque control, we realized that the reduction ratio due to the cable transmission between the hip pitch joint and respective motor is not taken in account in the Gearbox ratio (`Gearbox_M2J`), nor in the coupling matrices (`matrixM2J` and `matrixM2J`).

As you can see in https://github.com/robotology/QA/issues/256#issuecomment-336088281...

>Further details on the hip_pitch joint
>
>The hip_pitch joint has an additional cable transmission stage after the harmonic drive. The motor side pulley has a diameter of 50 [mm], while the joint side pulley has a diameter of 75 [mm], consequently, the additional reduction ration is: 75/50 =1.5

the reduction ratio due to the hip joint cable is 1.5, so we should get the following values for `matrixM2J`:
```
        <param name ="matrixM2J">
            0.667   0.000   0.000   0.000
            0.000   1.000   0.000   0.000
            0.000   0.000   1.000   0.000
            0.000   0.000   0.000   1.000
        </param>
```
the transformation from motor velocities to joint velocities being defined by:
$$
\dot{q} = T_{m2j} \\: G_{m2j} \\: \dot{\theta}
$$
where $T_{m2j}$ is the coupling matrix `matrixM2J`, $G_{m2j}$ the gearbox ratios, $\dot{q}$ and $\dot{\theta}$ respectively the vectors of joint and motor velocities.

Currently, `matrixM2J` is an identity and the gearbox ratio for the hip pitch is 100. So either we set the gearbox ratio to 150 either we update `matrixM2J` as shown above.

CC @randaz81 @valegagge @fiorisi @fabiobergonti @vibhoraggarwal @gabrielenava @traversaro @DanielePucci

DanielePucci, 2018-08-19:
If I am not wrong, this issue has an impact directly on the torque control because these matrices are used in the low level joint torque control. In this case, it may also justify some of the difficulties we had while performing torque control walking

CC
@S-Dafarra @gabrielenava @GiuliaP @hu-yue

nunoguedelha, 2018-08-19:
Yes exactly. And along with that the friction estimation results.

ale-git, 2018-08-20:
The reduction ratio was correctly 50/75 when the joint configuration was fixed in the source code in Controller.c (this part is now under #if 0):

```
    case eomc_ctrlboard_UPPERLEG:                //= 2,    //2FOC
    {
        Jjm = o->Jjm;
        Sjm = o->Sjm;
    
        Sjm[0][0] = 50.0f/75.0f;
        Jjm[0][0] = Sjm[0][0];
```

Evidently the 50/75 value has been omitted in the configuration files of the new version, erroneusly assuming that no coupling always implies identity matrix. Another way to correct the issue is leaving the identity matrix unchanged and change the Gearbox_M2J parameter from 100.0 to 150.0 in the same file.

ale-git, 2018-08-21:
The ratio doesn't affect the torque control, since the "motor" software entity under control is the "motor+gearbox" physical object seen as a single entity. It could have affected the robot if the iCub hip pitch motors were low level speed controlled like in R1, but they aren't. The only effect of the wrong gearbox ratio is in the joint velocity computed from the motor velocity, which isn't used in the controller (the derivative of the error in the PIDs is computed by locally differentiating the position error).

valegagge, 2018-08-21:
I prefer to update the gearbox ratio instead of the coupling matrix, because this is a reduction factor and not a couplig ratio. I know that the result is equal, but I think we should update the right parameter in order to mantain a clear information about the robot.

The gearbox parameters is the reduction factor from motor to joint, while the coupled matrix is used only when motors are coupled.

So please, could you update the `Gearbox_M2J` parameter to 150? 

After you'll check that new value is fine, we need to fix all robots with the same legs.
Could you advise me  if the change is ok, so I'll update others robots?
thanks!

nunoguedelha, 2018-08-22:
I actually don't agree with that for the following reasons:
- the coupling is supposed to define the transfer function (matrix) between a vector of [motor+physical gearbox] velocities and a vector of joint velocities,
- potencially any coupling system can have a "total" reduction (between sum of motor velocities and sum of joint velocities) ≠ 1,

so I don't think it's a good idea to consider that all the reduction factor in the "gearbox" ratio, it adds complexity and is error prone. For instance let's suppose some day we get to implement a low level control that models the elasticity of the cables, and depends on the reduction due to that cable (on the hip), we would have to do a workaround for retrieving the reduction factor from the "gearbox" ratio.

nunoguedelha, 2018-08-22:
Btw, currently all joints, coupled or not, are mapped to a coupling matrix, and it's more robust and simple to keep it that way. The less "exceptions" we use the better.

traversaro, 2018-08-22:
A nice reason to modify the coupling matrix instead of the `Gearbox_M2J` is that so we can keep the consistency between the `Gearbox_M2J` parameter and the reduction ratio of the harmonic drive.

valegagge, 2018-08-22:
`Gearbox_M2J` is the "total" reduction between motor to joint, not only harmonic drive.

There is a decument (written by to mechanicals guys) that specifies  each stage of reduction. It would be nice that these documenatation is sharebale.

traversaro, 2018-08-22:
> There is a decument (written by to mechanicals guys) that specifies each stage of reduction. It would be nice that these documenatation is sharebale.

Definitely, at least internally in IIT!

pattacini, 2019-12-13:
Any reason why we should keep this still open?

cc @nunoguedelha @traversaro

closed 2023-07-30 by pattacini