robotology/robots-configuration #125: Mismatch in the names of feet FT IMUs

prashanthr05, 2019-10-29:
While I was doing some base estimation experiments with the foot FT, I realized that there is a mismatch in the names of the foot FT IMUs added from the icub-model-generator https://github.com/robotology/icub-model-generator/pull/122 and those added at the robots-configuration level https://github.com/robotology/robots-configuration/pull/110

For example, considering the FT accelerometers, in the urdf model, these are added as `r_foot_ft_acc_3b14` while in the robots configurations, these are loaded with the names `r_lower_leg_ft_acc_3b14`.

We need to choose to make changes to one of the places to be coherent and minimize work. 
An alternative would be to set the proper metaData (`sensorName`, `frameName`, etc) through the  MAS interface.

Related issues: https://github.com/dic-iit/component_wholebody-teleoperation/issues/190

@nunoguedelha @traversaro @S-Dafarra @GiulioRomualdi @MiladShafiee @fjandrad

prashanthr05, 2019-10-29:
@traversaro  said:
> Ack, if you can come with a consistent naming proposal  that is great, otherwise I can try to push the issue in the iCub Tech backlog. In general, I suggest to keep an issue like this (in which multiple persons from different research lines can be interest) in a public repo, such as `icub-model-generator` or `robots-configuration`.

traversaro, 2019-10-29:
cc fyi @Nicogene that recently worked on this kind of sensors.

traversaro, 2019-10-29:
@nunoguedelha do you have any opinion on this, if we should make the icub-model-generator consistent with the  robots-configuration, or the other way around?

nunoguedelha, 2019-11-05:
@traversaro , considering tat the prefix in the name of the sensor should give an information about the support link of the sensor (e.g. `r_foot_ft_acc_xxx` is fixed on the foot), I suggest we keep **icub-model-generator** notation, and update  **robots-configuration** accordingly.

prashanthr05, 2019-12-05:
The support link is, technically, ankle. However, foot is more relatable than lower_leg, IMO! 

I have opened a PR here https://github.com/robotology/robots-configuration/pull/137

prashanthr05, 2020-01-30:
This was done in https://github.com/robotology/robots-configuration/pull/137

closed 2020-01-30 by prashanthr05