robotology/robots-configuration #146: Update all ETH robots to motion control ver 6

pattacini, 2019-12-20:
As per https://github.com/robotology/QA/issues/377 we need to update the following ETH robots to the use of `Motion Control Version 6`:
-  `iCubDarmstadt01`
- `iCubEdinburgh01`
- `iCubHeidelberg01`
- `iCubLausanne02`
- `iCubNancy01`
- `iCubNottingham01`
- `iCubSheffield01`
- `iCubSingapore01`
- `iCubTwente01`
- `iCubHamburg01`
 
cc @marcoaccame

marcoaccame, 2020-02-05:
here is an update.
at today in `devel` we have the following situation:

**robots** still to be to MCV 6 and ~robots~ recently migrated.
-  **iCubDarmstadt01**
- ~iCubEdinburgh01~
- ~iCubHeidelberg01~
- **iCubLausanne02**
- ~iCubNancy01~
- **iCubNottingham01**
- ~iCubSheffield01~
- ~iCubSingapore01~
- ~iCubTwente01~
- ~iCubHamburg01~

The migration of the remaining three robots is WIP.

marcoaccame, 2020-02-17:
all the above robots have migrated to mcv6 in `devel`.
see last three PRs: https://github.com/robotology/robots-configuration/pull/164, https://github.com/robotology/robots-configuration/pull/163 amd https://github.com/robotology/robots-configuration/pull/148

closed 2020-02-17 by marcoaccame