robotology/robots-configuration #159: Regression on devel branch with INSTALL_ALL_ROBOTS option enabled

traversaro, 2020-02-05:
~~~
2020-02-05T02:08:44.3657211Z -- Exists
2020-02-05T02:08:44.3657583Z CMake Error at /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/share/yarp/cmake/YarpInstallationHelpers.cmake:389 (file):
2020-02-05T02:08:44.3657716Z   file COPY cannot find
2020-02-05T02:08:44.3658072Z   "/home/runner/work/robotology-superbuild/robotology-superbuild/robotology/robots-configuration/iCubErzelli03/cartesian":
2020-02-05T02:08:44.3658195Z   No such file or directory.
2020-02-05T02:08:44.3658298Z Call Stack (most recent call first):
2020-02-05T02:08:44.3658390Z   iCubErzelli03/CMakeLists.txt:9 (yarp_install)
2020-02-05T02:08:44.3658458Z 
2020-02-05T02:08:44.3658506Z 
2020-02-05T02:08:44.3658885Z CMake Error at /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/share/yarp/cmake/YarpInstallationHelpers.cmake:389 (file):
2020-02-05T02:08:44.3659018Z   file COPY cannot find
2020-02-05T02:08:44.3659364Z   "/home/runner/work/robotology-superbuild/robotology-superbuild/robotology/robots-configuration/iCubErzelli03/estimators":
2020-02-05T02:08:44.3659486Z   No such file or directory.
2020-02-05T02:08:44.3659584Z Call Stack (most recent call first):
2020-02-05T02:08:44.3659686Z   iCubErzelli03/CMakeLists.txt:13 (yarp_install)
2020-02-05T02:08:44.3659739Z 
2020-02-05T02:08:44.3659783Z 
~~~

Failed robotology-superbuild job: https://github.com/robotology/robotology-superbuild/actions/runs/34879920 .
It would be handy to have a GitHub Actions to check this regression.

pattacini, 2020-02-05:
I'm gonna deal with this bug.

valegagge, 2020-02-05:
Hi @pattacini,
I get this error again. Could you reopen the issue and check again?

pattacini, 2020-02-05:
Which branch did you build?
`devel` builds fine on my system.

valegagge, 2020-02-05:
I'm on `master`

pattacini, 2020-02-05:
@valegagge fixed in `master` too 👍

Given that #157 had `devel` as the base branch, this must have been a longstanding bug 🐞

closed 2020-02-05 by pattacini

valegagge, 2020-02-05:
@pattacini thanks very much!