robotology/robots-configuration #171: Update wiki and templates

vtikha, 2020-03-02:
@valegagge pointed out that we should update the [manual](http://wiki.icub.org/wiki/Manual#Ten._Robot_configuration) containing the robots configuration information and the [templates](https://github.com/robotology/robots-configuration/tree/master/iCubTemplates)

cc @marcoaccame

traversaro, 2021-11-12:
Just for reference, this is indeed quite precious information on the configuration of the low level boards (including for example documentation for each parameter of the xml file corresponding to each board, for example https://github.com/robotology/robots-configuration/blob/f3fef4277717ae87d825a9a46d994ccd21fef3ec/iCubTemplates/iCubTemplateV3_0/hardware/electronics/body_part--ebX-jA_B-eln.xml#L63 https://github.com/robotology/robots-configuration/blob/f3fef4277717ae87d825a9a46d994ccd21fef3ec/iCubTemplates/iCubTemplateV3_0/hardware/electronics/pc104.xml#L28 and https://github.com/robotology/robots-configuration/blob/f3fef4277717ae87d825a9a46d994ccd21fef3ec/iCubTemplates/iCubTemplateV3_0/hardware/FT/body_part-ebX-strain.xml#L70) that is not easy to find for users of the robot. Today I showed this doc to @GiulioRomualdi and @isorrentino that had no idea about it.

valegagge, 2021-11-26:
Hi,
I'd like to point out that now we are using the configuration file version 6, but the templates have not been updated. 
Could we update them?

@traversaro: Maybe we could add the information of the templates in the readme file in order to make such information more accessible to everybody.

traversaro, 2021-11-26:
> @traversaro: Maybe we could add the information of the templates in the readme file in order to make such information more accessible to everybody.

Good idea! Probably it also make sense to link it somehow from https://github.com/icub-tech-iit/documentation .

pattacini, 2021-12-02:
Added to our next sprint.

pattacini, 2022-01-08:
Rescheduled at the beginning of PI10.

pattacini, 2022-03-19:
The work has been done in:
- #345 

Regarding

>Good idea! Probably it also make sense to link it somehow from https://github.com/icub-tech-iit/documentation.

This is a good point that will be addressed by @valegagge in a second step.

closed 2022-03-19 by pattacini