robotology/robots-configuration #177: Nightly merge keeps failing in fork

S-Dafarra, 2020-03-23:
We forked ``robots-configuration`` in our organization (https://github.com/dic-iit/robots-configuration), but the GitHub action "Nightly Merge" (https://github.com/dic-iit/robots-configuration/actions) fails every time.

I thought if modifying or removing this action, but then there would have been a discrepancy between this master branch and the one in my organization, complicating future PRs.

How can I deal with this?

S-Dafarra, 2020-03-23:
cc @pattacini @traversaro

pattacini, 2020-03-23:
The action is failing because of the token we use that is `marcoaccame_TOKEN`. 

As this action operates on the couple `master`/`devel`, you could disable it straight from the dashboard of your fork https://github.com/dic-iit/robots-configuration/settings/actions.

S-Dafarra, 2020-03-23:
Clear! I did not notice that option.
Thanks a lot @pattacini

closed 2020-03-23 by S-Dafarra