robotology/robots-configuration #186: iCubGenova02 not starting after #178

lrapetti, 2020-06-16:
After #178 was merged, the startup is failing. It is correctly started by moving back to previous https://github.com/robotology/robots-configuration/commit/1a05e7825afcab6ade8a9477b56f584c95a18ccf.

I think the problem is due to the fact that the Firmware of the robot FTs is not updated.
CC @marcoaccame 

Here is the output:
```
/iCubGenova02(devel)$ yarprobotinterface --config icub_wbd.xml 
[DEBUG] Reading file /usr/local/src/robot/robotology-superbuild/robotology/robots-configuration/iCubGenova02/icub_wbd.xml
[DEBUG] yarprobotinterface: using xml parser for DTD v3.x
[DEBUG] Reading file /usr/local/src/robot/robotology-superbuild/robotology/robots-configuration/iCubGenova02/icub_wbd.xml
[DEBUG] Preprocessor complete in:  0.0375211 s
[INFO] |yarp.os.Port| Port /icub/yarprobotinterface active at tcp://10.0.0.2:10011/
[INFO] startup phase starting...
[DEBUG] eth::parser::print(pc104Data) for PC104:
[DEBUG] PC104/PC104IpAddress:PC104IpPort =  10.0.1.104:12345
[DEBUG] PC104/PC104TXrate =  1
[DEBUG] PC104/PC104RXrate =  5
[DEBUG] EthSender is a PeriodicThread with txrate = 1 ms
[DEBUG] EthReceiver is a PeriodicThread with rxrate = 5 ms
[WARNING] in EthReceiver::config() the config socket has queue size =  2097152 ; you request ETHRECEIVER_BUFFER_SIZE= 
[DEBUG] eth::parser::print(boardData) for BOARD head-eb20-j0_1
[DEBUG] ETH_BOARD/ETH_BOARD_PROPERTIES:
[DEBUG] ETH_BOARD/ETH_BOARD_PROPERTIES/IpAddress =  10.0.1.20
[DEBUG] ETH_BOARD/ETH_BOARD_PROPERTIES/IpPort =  12345
[DEBUG] ETH_BOARD/ETH_BOARD_PROPERTIES/Type =  mc4plus
[DEBUG] ETH_BOARD/ETH_BOARD_PROPERTIES/maxSizeRXpacket =  768
[DEBUG] ETH_BOARD/ETH_BOARD_PROPERTIES/maxSizeROP =  384
[DEBUG] ETH_BOARD/ETH_BOARD_SETTINGS:
[DEBUG] ETH_BOARD/ETH_BOARD_SETTINGS/Name =  head-eb20-j0_1
[DEBUG] ETH_BOARD/ETH_BOARD_SETTINGS/RUNNINGMODE/(period, maxTimeOfRXactivity, maxTimeOfDOactivity, maxTimeOfTXactivity, TXrateOfRegularROPs) =  1000 400 300 300 5
[DEBUG] ETH_BOARD/ETH_BOARD_ACTIONS/MONITOR_ITS_PRESENCE
[DEBUG] ETH_BOARD/ETH_BOARD_ACTIONS/MONITOR_ITS_PRESENCE/(enabled, timeout, periodOfMissingReport) =  true 0.02 60
[DEBUG] eth::EthMonitorPresence::config(): monitoring of presence is ON for BOARD 10.0.1.20 (head-eb20-j0_1) with timeout = 0.02 sec and period of missing report = 60 sec
[DEBUG] TheEthManager::requestResource2(): has just succesfully created a new EthResource for board of type mc4plus with IP =  10.0.1.20
[WARNING] embObjMC BOARD  head-eb20-j0_1 Missing OTHER_CONTROL_PARAMETERS.DeadZone parameter. I'll use default value. (see documentation for more datails)
[DEBUG] EthResource::verifyBoardPresence() found BOARD head-eb20-j0_1 with IP 10.0.1.20 after 0.00337386 seconds
[DEBUG] EthResource::verifyBoardTransceiver() has validated the transceiver of BOARD head-eb20-j0_1 with IP 10.0.1.20
[DEBUG] EthResource::cleanBoardBehaviour() has cleaned the application in BOARD head-eb20-j0_1 with IP 10.0.1.20 : config mode + cleared all its regulars
[DEBUG] EthResource::setTimingOfRunningCycle() for BOARD head-eb20-j0_1 with IP 10.0.1.20 has succesfully set: cycletime = 1000 usec, RX DO TX = ( 400 300 300 ) usec and TX rate = 5  every cycle
[INFO] EthResource::askBoardVersion() found BOARD head-eb20-j0_1 @ IP 10.0.1.20 of type mc4plus with FW = ver 3.23 built on 2019 Dec 13 16:30
[DEBUG]  from BOARD 10.0.1.20 (head-eb20-j0_1), src LOCAL, adr 0, time 72s 237m 339u: (code 0x05000015, par16 0x2000 par64 0x0000000000000000) -> CFG: EOtheEncoderReader can be correctly configured. p16&0xf000: number of joint; primary encs: failure mask in p16&0x000f and errorcodes in p64&0x0000ffff; secondary encs: failure mask in p16&0x00f0 and errorcodes in p64&0xffff0000 + .
[DEBUG]  from BOARD 10.0.1.20 (head-eb20-j0_1), src LOCAL, adr 0, time 72s 238m 404u: (code 0x05000013, par16 0x0000 par64 0x0000000000000000) -> CFG: EOtheMotionController can correctly configure mc4plus-based motion. more info will follow + .
[DEBUG] embObjMotionControl:serviceVerifyActivate OK!
[INFO] created device <embObjMotionControl>. See C++ class yarp::dev::embObjMotionControl for documentation.
[DEBUG] eth::parser::print(boardData) for BOARD head-eb21-j2_5
[DEBUG] ETH_BOARD/ETH_BOARD_PROPERTIES:
[DEBUG] ETH_BOARD/ETH_BOARD_PROPERTIES/IpAddress =  10.0.1.21
[DEBUG] ETH_BOARD/ETH_BOARD_PROPERTIES/IpPort =  12345
[DEBUG] ETH_BOARD/ETH_BOARD_PROPERTIES/Type =  mc4plus
[DEBUG] ETH_BOARD/ETH_BOARD_PROPERTIES/maxSizeRXpacket =  768
[DEBUG] ETH_BOARD/ETH_BOARD_PROPERTIES/maxSizeROP =  384
[DEBUG] ETH_BOARD/ETH_BOARD_SETTINGS:
[DEBUG] ETH_BOARD/ETH_BOARD_SETTINGS/Name =  head-eb21-j2_5
[DEBUG] ETH_BOARD/ETH_BOARD_SETTINGS/RUNNINGMODE/(period, maxTimeOfRXactivity, maxTimeOfDOactivity, maxTimeOfTXactivity, TXrateOfRegularROPs) =  1000 400 300 300 5
[DEBUG] ETH_BOARD/ETH_BOARD_ACTIONS/MONITOR_ITS_PRESENCE
[DEBUG] ETH_BOARD/ETH_BOARD_ACTIONS/MONITOR_ITS_PRESENCE/(enabled, timeout, periodOfMissingReport) =  true 0.02 60
[DEBUG] eth::EthMonitorPresence::config(): monitoring of presence is ON for BOARD 10.0.1.21 (head-eb21-j2_5) with timeout = 0.02 sec and period of missing report = 60 sec
[DEBUG] TheEthManager::requestResource2(): has just succesfully created a new EthResource for board of type mc4plus with IP =  10.0.1.21
[WARNING] embObjMC BOARD  head-eb21-j2_5 Missing OTHER_CONTROL_PARAMETERS.DeadZone parameter. I'll use default value. (see documentation for more datails)
[DEBUG] EthResource::verifyBoardPresence() found BOARD head-eb21-j2_5 with IP 10.0.1.21 after 0.00357771 seconds
[DEBUG] EthResource::verifyBoardTransceiver() has validated the transceiver of BOARD head-eb21-j2_5 with IP 10.0.1.21
[DEBUG] EthResource::cleanBoardBehaviour() has cleaned the application in BOARD head-eb21-j2_5 with IP 10.0.1.21 : config mode + cleared all its regulars
[DEBUG] EthResource::setTimingOfRunningCycle() for BOARD head-eb21-j2_5 with IP 10.0.1.21 has succesfully set: cycletime = 1000 usec, RX DO TX = ( 400 300 300 ) usec and TX rate = 5  every cycle
[INFO] EthResource::askBoardVersion() found BOARD head-eb21-j2_5 @ IP 10.0.1.21 of type mc4plus with FW = ver 3.23 built on 2019 Dec 13 16:30
[DEBUG]  from BOARD 10.0.1.21 (head-eb21-j2_5), src LOCAL, adr 0, time 72s 386m 352u: (code 0x05000015, par16 0x4000 par64 0x000000000000ff00) -> CFG: EOtheEncoderReader can be correctly configured. p16&0xf000: number of joint; primary encs: failure mask in p16&0x000f and errorcodes in p64&0x0000ffff; secondary encs: failure mask in p16&0x00f0 and errorcodes in p64&0xffff0000 + .
[DEBUG]  from BOARD 10.0.1.21 (head-eb21-j2_5), src LOCAL, adr 0, time 72s 387m 439u: (code 0x05000013, par16 0x0000 par64 0x0000000000000000) -> CFG: EOtheMotionController can correctly configure mc4plus-based motion. more info will follow + .
[DEBUG] embObjMotionControl:serviceVerifyActivate OK!
[INFO] created device <embObjMotionControl>. See C++ class yarp::dev::embObjMotionControl for documentation.
[ERROR] |yarp.devices.ControlBoardWrapper| Parameter networks use deprecated syntax
[... 779 lines cut ...]
[INFO] |yarp.os.Port| Port /icub/cartesianController/right_arm/events:o active at tcp://10.0.0.2:10080/
[INFO] |yarp.os.Port| Port /icub/cartesianController/right_arm/rpc:i active at tcp://10.0.0.2:10081/
[INFO] created device <cartesiancontrollerserver>. See C++ class ServerCartesianController for documentation.
[WARNING] There was some problem opening one or more devices. Please check the log and your configuration
[ERROR] One or more devices failed opening... see previous log messages for more info
[ERROR] yarprobotinterface intercepted a segmentation fault caused by a faulty plugin:
[ERROR] /usr/local/src/robot/icub-main/build/lib/iCub/cartesiancontrollerserver.so(_ZN25ServerCartesianController9detachAllEv+0x40) [0x7f7b64358b80]
Trace requested at /usr/local/src/robot/robotology-superbuild/robotology/YARP/src/yarprobotinterface/Module.cpp:86 by code called from:
/usr/local/src/robot/robotology-superbuild/build/install/lib/libYARP_os.so.3(_Z16yarp_print_traceP8_IO_FILEPKcj+0x3b) [0x7f7bf5b15aab]
yarprobotinterface(+0x614a) [0x56085240a14a]
/lib/x86_64-linux-gnu/libc.so.6(+0x3ef20) [0x7f7bf514cf20]
/usr/local/src/robot/icub-main/build/lib/iCub/cartesiancontrollerserver.so(_ZN25ServerCartesianController9detachAllEv+0x40) [0x7f7b64358b80]
/usr/local/src/robot/icub-main/build/lib/iCub/cartesiancontrollerserver.so(_ZN25ServerCartesianController5closeEv+0x1a) [0x7f7b64362cfa]
[... cut ...]
/usr/local/src/robot/robotology-superbuild/build/install/lib/libYARP_robotinterface.so.3(_ZN4yarp14robotinterface5Robot10enterPhaseENS0_11ActionPhaseE+0x29d4) [0x7f7bf6087584]
yarprobotinterface(+0x8601) [0x56085240c601]
/usr/local/src/robot/robotology-superbuild/build/install/lib/libYARP_os.so.3(_ZN4yarp2os8RFModule9runModuleERNS0_14ResourceFinderE+0x72) [0x7f7bf5b5f8e2]
yarprobotinterface(+0x5110) [0x560852409110]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7) [0x7f7bf512fb97]
yarprobotinterface(+0x51ea) [0x5608524091ea]
Segmentation fault (core dumped)

```

julijenv, 2020-06-16:
Ciao @lrapetti ,
coudl you please remove the entire log file you add to the issue and put everything in one file . txt, it is way more readable that way.
thx in advance

marcoaccame, 2020-06-17:
yes, the firmware of the robot is not updated. 
in particular the log tells that the required fw version of strain2 board is 2.0.7 and there is still 2.0.4.

```
[INFO] - STRAIN of type strain2 named id_l_upper_leg_strain @ CAN2:13 
 with required protocol version = (2, 0) and required firmware version = (2, 0, 7)
[ERROR]  from BOARD 10.0.1.6 (left_leg-eb6-j0_3) @ 81s 520m 547u: CAN discovery 
has detected 1 invalid eobrd_strain2 boards in CAN2:
[WARNING]  from BOARD 10.0.1.6 (left_leg-eb6-j0_3) @ 81s 520m 967u: CAN discovery has 
detected a eobrd_strain2 board in CAN2 addr 13 with 
can protocol ver 2.0 and application ver 2.0.4 Search time was 2 ms
[ERROR] 1 of 1: wrong eobrd_strain2 BOARD 10.0.1.6:CAN2:13 because it has:  
WRONG APPLICATION VERSION 
```

the required fw version for every board in the robot are in file: https://github.com/robotology/icub-firmware-build/blob/master/info/firmware.info.xml), and instructions on how to update are in https://github.com/robotology/icub-firmware-build/tree/master/docs

lrapetti, 2020-06-17:
> yes, the firmware of the robot is not updated.

Thanks @marcoaccame. 

Some of the main `iCubGenova02` users might be using old configuration files for whom the current firmware is working. I don't want to brake their applications.
I think @marco-monforte @GiuliaDAngelo (with the other `iCubGenova02` users) can decide wether/when to proceed with the update of the robot firmware.

marco-monforte, 2020-06-17:
Hi everyone, I'll check that we don't have any issue in updating the firmware and I'll do it as soon as possible.

marco-monforte, 2020-06-30:
Hey guys, the firmware has been updated and now the strain2 has the 2.0.7 version.

Thanks for the help @marcoaccame .

@lrapetti when you want, we can try what you were doing and close the issue hopefully.

lrapetti, 2020-06-30:
Thanks @marco-monforte @marcoaccame, we tried to run the robot with the configuration file in master ([`icub_wbd.xml`](https://github.com/robotology/robots-configuration/blob/master/iCubGenova02/icub_wbd.xml)) and it started correctly.

We can close the issue.

closed 2020-06-30 by lrapetti