robotology/robots-configuration #222: iCubGenova09 yarprobotinterface throws error "Parameter networks use deprecated syntax"

S-Dafarra, 2020-12-14:
Every time we run the ``yarprobotinterface`` we get the not fatal error
```
Parameter networks use deprecated syntax
```

It seems it is thrown by https://github.com/robotology/yarp/blob/9ce40a416ce872154c90130a04648eb6923eb536/src/devices/ControlBoardWrapper/ControlBoardWrapper.cpp#L489, but I am not able to spot what causes this issue in the configuration files, nor how to avoid that.

S-Dafarra, 2020-12-14:
I suppose that lines like 
https://github.com/robotology/robots-configuration/blob/a1605113ce72472c3442a8d72bfbe7d97ec7152b/iCubGenova09/wrappers/motorControl/left_leg-mc_wrapper.xml#L8
should be changed to 
```xml
<elem name="FirstSetOfJoints">  (0  1  0  1) </elem> 
```
but the documentation does not seem to be in this way
https://github.com/robotology/yarp/blob/yarp-3.4/src/devices/ControlBoardWrapper/ControlBoardWrapper.h#L171-L175

Then, I am wondering if this is a problem of ``yarp`` or of the configuration files. This problem was also spotted in https://github.com/robotology/icub-main/issues/661

pattacini, 2020-12-14:
Thanks @S-Dafarra for reporting

I believe that your analysis is correct 👍🏻 

`(0  1  0  1)` is the new preferred way of passing parameters, whereas we have still `0  1  0  1` available as a fallback (see https://github.com/robotology/yarp/blob/9ce40a416ce872154c90130a04648eb6923eb536/src/devices/ControlBoardWrapper/ControlBoardWrapper.cpp#L490-L493).

As a result, the documentation https://github.com/robotology/yarp/blob/yarp-3.4/src/devices/ControlBoardWrapper/ControlBoardWrapper.h#L171-L175 seems outdated.

S-Dafarra, 2020-12-14:
Actually, it seems that this error was reintroduced in https://github.com/robotology/yarp/commit/5da1cf3c02d288a24470aad222e2ac127dc370a9 when changing the log mechanism in ``yarp``. Before, that line was commented since ages

S-Dafarra, 2020-12-14:
Actually, that error was commented out since the first time it was pushed https://github.com/robotology/yarp/commit/d164b21ba83c85cdcd8aafb905d6ec1531439a4f

pattacini, 2020-12-14:
Deprecation should be signaled as warnings rather than errors, I suppose.
Anyway, would you be willing to test the fix in the conf files with `iCubGenova09` @S-Dafarra ?

Later on, we will be responsible for updating the remaining robots.

S-Dafarra, 2020-12-14:
At this point I suppose it is no more a problem of the configuration files, but rather of ``yarp``. In particular, it is not clear if that syntax has ever been deprecated.

> Anyway, would you be willing to test the fix in the conf files with iCubGenova09 @S-Dafarra ?

I will not have access to the robot for a couple of days I guess :wink:

pattacini, 2020-12-14:
>it is not clear if that syntax has ever been deprecated.

cc @drdanz. In the meanwhile, I can try to track it back through the announcements.

>I will not have access to the robot for a couple of days I guess 😉

You're right! @davidetome could deal with this at the end of head replacement.

pattacini, 2020-12-14:
@drdanz it seems that we went from 
```c++
 // yError<<"Parameter networks use deprecated syntax\n";
```
to
```c++
yCError(CONTROLBOARDWRAPPER) << "Parameter networks use deprecated syntax";
```
via https://github.com/robotology/yarp/commit/5da1cf3c02d288a24470aad222e2ac127dc370a9 without highlighting the change in YARP 3.4.x announcements (at least I didn't find it).

Thus, I'd propose that we:
- change `yCError` into `yCWarning`
- update the inline documentation at https://github.com/robotology/yarp/blob/yarp-3.4/src/devices/ControlBoardWrapper/ControlBoardWrapper.h#L171-L175
- we update robots conf files.

S-Dafarra, 2020-12-14:
Sounds good to me!

pattacini, 2020-12-17:
@julijenv did sort our this already for `iCubWaterloo01` in #223 ✔

pattacini, 2020-12-24:
- Conf files updated in #227.
- PR on YARP opened in https://github.com/robotology/yarp/pull/2446.

closed 2020-12-24 by pattacini