robotology/robots-configuration #228: Wrong coefficients coupling the eyes' vergence?

pattacini, 2021-01-11:
Given that the vergence `Vg` amounts to the difference between the single eyes' joints `L-R` (see the [wiki][1]), I think that the [values contained in the eyes' coupling matrices][2] should be **`-1.0 1.0`** instead of `-0.5  0.5`.

Differently, as `Vs = (L + R) / 2` it turns out that the line above reporting `0.5  0.5` is spot on.

Can this reasoning be assumed correct?

This is somehow linked to https://github.com/robotology/icub-firmware/issues/71.

cc @davidetome @randaz81 @ale-git @traversaro @plinioMoreno 

[1]: https://wiki.icub.org/wiki/Vergence,_Version_and_Disparity
[2]: https://github.com/robotology/robots-configuration/blob/master/iCubGenova04/hardware/mechanicals/head-eb21-j2_5-mec.xml#L45

pattacini, 2021-01-18:
Any comments on this @ale-git @randaz81 ?

pattacini, 2021-01-19:
Discussed this with @randaz81 today, who agrees with what said above.
However, the wrong matrix coefficients get subsumed by the PID gains, hence there ain't actually any real control problem.

Given that it's more correct to have `-1.0 1.0`, it'd be worth updating the conf files anyway.

pattacini, 2021-03-02:
Apparently, also matrixE2J should contain coupling terms!

pattacini, 2023-07-28:
/remind September 03 let's address this once for good.

pattacini, 2023-09-18:
Link in the OP fixed.

pattacini, 2023-09-18:
To sum up, here's the list of things that we need to do:
- Stick to one iCub with eyes; most likely iCubGenova11.
- Update iCubGenova11 following these [instructions](https://github.com/robotology/robots-configuration/issues/228#issue-783526278).
- Test that the eyes will still move correctly w/o big overshoot.
- Update all the remaining maintained robots.

martinaxgloria, 2023-09-27:
Hi @pattacini,

as you suggested, I booked iCubGenova11 to do the tests with the new values of eyes' vergence. I took a video of the eyes' movement after the upload:

https://github.com/robotology/robots-configuration/assets/114698424/1adeb6c5-f5f3-4a80-90ab-dce5e693c6f5

I have little experience with it, but the movement seems smooth, also compared to the previous version with the _wrong_ vergence values:

https://github.com/robotology/robots-configuration/assets/114698424/016518b5-3a1f-4f85-969a-ff8347b84727

Do you agree? If yes, I'll proceed with the update of all the maintained robots.

martinaxgloria, 2023-09-27:
Just to clarify, I only update [this line](https://github.com/robotology/robots-configuration/blob/master/robots-icebox/iCubGenova04/hardware/mechanicals/head-eb21-j2_5-mec.xml#L45) with `-1.0 1.0` instead of `-0.5 0.5`.

pattacini, 2023-09-27:
Great, @martinaxgloria 👍🏻 

That's all correct!

The videos are nice. However, given the control context, I would expand on the validation phase by including some data acquisitions of the commands and joint feedback. These traces would help us appreciate small details that are not visible in the videos, like overshoots, rising times...

martinaxgloria, 2023-09-27:
You're right @pattacini! I'll do the acquisitions and I'll report here the results.

pattacini, 2023-09-27:
You can talk to @davidetome and/or @mfussi66, who do have some code snippets for logging both commands and feedback you may consider reusing.

martinaxgloria, 2023-10-02:
Today I plotted the data I've acquired on iCubGenova11 after the change in the vergence coupling matrix and here they are:

[image]

I hope that these quantities are the ones that @pattacini referred to. In particular, I plotted the `encoder`, `motor encoder`, `pwm` and `current` values for the `eyes_tilt`, `eyes_vers` and `eyes_verg` joints. These values were obtained by setting a sequence from the yarpmotorgui from the home position, to the upper limit, then the lower limit and then home again for the three eyes' joints. I set a time of 5 sec for each movement (e.g. from home to upper limit in 5 sec). Then I start the acquisition and I ran the code to save the values in order to plot them at the end.

The only particular thing that stands out is the peak of current and duty cycle percentage around the second 45 that corresponds to the `eyes_verg` joint reaching its upper limit. In your opinion, is it consistent with the movement?

cc @pattacini @mfussi66

pattacini, 2023-10-02:
Great that you managed to plot these results 👍🏻 
A couple of important points though:
- The joint references are missing.
- The 5-second movements are far slower than the dynamics we're interested in with the eyes. Eyes need to move much much faster, meaning fractions of a second. For example, let's stick to 0.3 s.

martinaxgloria, 2023-10-04:
Yesterday I repeated the tests acquiring the joint references and the other quantities with much faster movements (0.3 sec per movement). To make a comparison, I also do the tests and the acquisitions before the change:

[image]
and the comparison between the joint encoder angles and the reference values:

[image]

After the changes, instead:

[image]

[image]

@mfussi66 noticed that the sampling time it's high (when I did again the acquisitions maybe I accidentally change this parameter and I didn't notice it). So far, I decided to post here the results, but I'll do again the acquisitions and the plots with more samplings to have a more accurate analysis.

cc @pattacini

pattacini, 2023-10-04:
Overall, I'm happy with the results and for me it's green light 🟢
Thus, we can proceed with the update of `robots-configuration`.

> @mfussi66 noticed that the sampling time it's high (when I did again the acquisitions maybe I accidentally change this parameter and I didn't notice it). So far, I decided to post here the results, [...]

It seems so, yeah. However, what was the sample time you used?

martinaxgloria, 2023-10-04:
> It seems so, yeah. However, what was the sample time you used?

I computed it from the acquired samples and it is 0.1s

pattacini, 2023-10-04:
> > It seems so, yeah. However, what was the sample time you used?
> 
> I computed it from the acquired samples and it is 0.1s

Ok, it's fairly low, but the graphs above demonstrate that the reference is well-tracked anyhow.

For the record, good a sample time when dealing with control issues is 10 ms at the YARP layer.

martinaxgloria, 2023-10-04:
Yes, thanks for the clarification @pattacini!
So, can I proceed with the update of robots-configuration or it's better to re-do the acquisitions with the right sample time before?

pattacini, 2023-10-04:
>So, can I proceed with the update of robots-configuration or it's better to re-do the acquisitions with the right sample time before?

Let's proceed with robots-configuration 👍🏻

martinaxgloria, 2023-10-04:
PR here:

- https://github.com/robotology/robots-configuration/pull/574