robotology/robots-configuration #233: Clarification on syntax issue after commit a1cf818

xEnVrE, 2021-01-26:
This morning I tried to run iCubGenova01 after git pulling the latest `robots-configuration` and I could not run the robot due to the following issue (reporting from the output of `yarprobotinterface`)

```
2,278432 <DEBUG> (device virtualAnalogServer) (name "/icub/joint_vsens/left_arm:i") (net_VFT_LA (0 5 0 5)) (networks (net_VFT_LA)) (period 10) (robotName icub) 
 2,278554 <DEBUG> Using VirtualAnalogServer 
 2,278605 <ERROR> Check network parameters in part description. I was expecting net_VFT_LA followed by four integers 
 2,278655 <ERROR> Driver <virtualAnalogServer> was found but could not open 
 2,278716 <WARNING> Cannot open device left_arm_VFTserver 
 2,278753 <WARNING> Cannot open device left_arm_VFTserver 
 2,278783 <DEBUG> (device virtualAnalogServer) (name "/icub/joint_vsens/right_arm:i") (net_VFT_RA (0 5 0 5)) (networks (net_VFT_RA)) (period 10) (robotName icub) 
 2,278827 <DEBUG> Using VirtualAnalogServer 
 2,278864 <ERROR> Check network parameters in part description. I was expecting net_VFT_RA followed by four integers 
 2,278901 <ERROR> Driver <virtualAnalogServer> was found but could not open 
 2,278937 <WARNING> Cannot open device right_arm_VFTserver 
 2,278966 <WARNING> Cannot open device right_arm_VFTserver 
 2,278994 <DEBUG> (device virtualAnalogServer) (name "/icub/joint_vsens/torso:i") (net_VFT_TO (0 5 0 5)) (networks (net_VFT_TO)) (period 10) (robotName icub) 
 2,279033 <DEBUG> Using VirtualAnalogServer 
 2,279067 <ERROR> Check network parameters in part description. I was expecting net_VFT_TO followed by four integers 
 2,279109 <ERROR> Driver <virtualAnalogServer> was found but could not open 
 2,279144 <WARNING> Cannot open device torso_VFTserver 
 2,279173 <WARNING> Cannot open device torso_VFTserver 
```

After checking I found that the issue, at least on our setup, is coming from https://github.com/robotology/robots-configuration/commit/a1cf818f6fa3f88a6358d4829617507d95613480.

Specifically, it seems that the parentheses surrounding the integers are not recognized by the parser

https://github.com/robotology/robots-configuration/blob/2bc1ca0be022a02a42f87d2646a205af06d5eff0/iCubGenova01/wrappers/VFT/left_arm_VFT_wrapper.xml#L6

https://github.com/robotology/robots-configuration/blob/2bc1ca0be022a02a42f87d2646a205af06d5eff0/iCubGenova01/wrappers/VFT/right_arm_VFT_wrapper.xml#L6

https://github.com/robotology/robots-configuration/blob/2bc1ca0be022a02a42f87d2646a205af06d5eff0/iCubGenova01/wrappers/VFT/torso_VFT_wrapper.xml#L6

I wanted to ask the following: is this change wanted and I need to update `YARP` and/or `icub-main` because the parser was changed or is there something else that I am missing?

Thank you!

xEnVrE, 2021-01-26:
It seems that, YARP-side, the error is thrown here:

https://github.com/robotology/yarp/blob/b418c1b37ab7e317369e64f2840bfad3881ad7f5/src/devices/VirtualAnalogWrapper/VirtualAnalogWrapper.cpp#L150

traversaro, 2021-01-26:
Thanks for raising the point @xEnVrE . For reference, the related PR I guess is https://github.com/robotology/robots-configuration/pull/227 . 

The problem seems that the virtualAnalogServer (see https://github.com/robotology/yarp/blob/0fcdd28793d2b8d2356c72a2edb728774a9ef7a3/src/devices/VirtualAnalogWrapper/VirtualAnalogWrapper.cpp#L154 ) was never updated to support the same networks format used by the controlboardwrapper2 (see https://github.com/robotology/yarp/blob/b418c1b37ab7e317369e64f2840bfad3881ad7f5/src/devices/ControlBoardWrapper/ControlBoardWrapper.cpp#L466), and it is only supporting the legacy format (that the controlboard marked deprecated, see https://github.com/robotology/yarp/blob/b418c1b37ab7e317369e64f2840bfad3881ad7f5/src/devices/ControlBoardWrapper/ControlBoardWrapper.cpp#L489).

While the quickest workaround is probably to revert the changes that https://github.com/robotology/robots-configuration/pull/227 applied to the `virtualAnalogServer` devices files (it was supposed to only affect `controlboardwrapper2` devices files), the cleanest change for the future is probably to also modify `virtualAnalogServer` to support exactly the same formats supported by the `controlboardwrapper2` device.

pattacini, 2021-01-26:
Thanks @xEnVrE and @traversaro for spotting this!

I think the best solution is that we update the `virtualAnalogServer` asap. We can do that!
Changing backward and forward lots of files in the repo is unwieldy.

Stay tuned.

xEnVrE, 2021-01-26:
Thank you @pattacini

traversaro, 2021-01-26:
> I think the best solution is that we update the `virtualAnalogServer` asap. We can do that!

This is probably something that we need to monitor for 2021.02 @Nicogene .

pattacini, 2021-01-26:
@xEnVrE you can continue to work on the local copy of `robots-configuration` by reverting the changes in #227 anyway.
We will make the patch land within tomorrow.

xEnVrE, 2021-01-26:
Absolutely, I changed manually the three lines in order to run the robot.

Thank you

pattacini, 2021-01-26:
PR opened up in https://github.com/robotology/yarp/pull/2467.

cc @triccyx

pattacini, 2021-01-28:
@xEnVrE the PR has been just merged!
If you use `yarp@yarp-3.4` you should be fine.

closed 2021-01-28 by pattacini