robotology/robots-configuration #247: Request for Comments: Clarify location of robot-specific manually added files?

traversaro, 2021-03-24:
Historically, the robots-configuration has been used to host two kind of files:
* The "official" (with a lot of quotes : ) ) configuration files for a given robot, that are typically maintained by the iCub Tech, and that routinely need to be massively updated for several reason (bugs found, change in the software that requires update in the configuration files). This files are not exactly the same for two different (even if identical) robots, due to calibration settings and similar robot specific parameters, but at least their structure is (or should be) consistent  across robot of the same version. 
* Robot-specific configuration and material: this includes robot-specific material that is maintained by specific Research Lines that used the robot (in IIT) or robot groups. While one school of thought is that this material should be in repositories that are maintained by the groups that are responsible for them, in practice for configuration files that are part of the "robot setup" it is quite convenient to be able to managed them with the rest of the robot configuration files.

This is de-facto situation. However, I think that a possible improvement (especially to simplify the possibility of doing **automatic changes** to the "official" configuration files) is to **dedicate a given directory in the robot directory for robot-specific files**, something called `extra` or similar ([bikeshedding](https://www.youtube.com/watch?v=5ouopR_eG08) on the name is welcome), in which all the robot-specific files can be placed. In this way, it would be easy to separate between "official" iCub configuration files and robot-specific ones.

traversaro, 2021-03-24:
Comments on this proposal are welcome: @Nicogene @pattacini @marcoaccame @S-Dafarra @violadelbono @davidetome @triccyx (feel free to tag more people that could be interesting).

pattacini, 2021-03-24:
Good suggestion @traversaro 👍🏻 
Also, `extra` sounds good.

traversaro, 2021-03-24:
>  Also, `extra` sounds good.

I must confessed that I borrowed the concept from the FMI3 standard, see [Section 2.5.1.6. Directory Extra](https://fmi-standard.org/docs/3.0-dev/#structure-of-zip).

S-Dafarra, 2021-03-24:
Sounds good. Among the files that could go in that directory there would be also the ``yarpmanager`` application files for example (since those are usually modified to consider the specific machines in the network for example). Hence this repo would store the configuration files in a broader sense, not just related to the robot itself, but to the entire setup.

Nicogene, 2021-03-24:
I like the idea, the `extra` folder could contain all the "extern" configuration files stored with the contexts hierarchy

pattacini, 2021-03-24:
@Nicogene if you mean the .ini files, that's not the case!
.ini files shall be in the robot root directory to make all the resource-finder pipeline work as intended.

pattacini, 2021-04-01:
So, go ahead @S-Dafarra as agreed!

closed 2021-04-01 by pattacini