robotology/robots-configuration #280: Update joint limits iCub3

AlexAntn, 2021-07-12:
After work done on the iCub3 gazebo model, submitted in [this PR](https://github.com/robotology/icub-models-generator/pull/202), we have updated the **upper limit for the shoulder roll, from 160 to 163**.

This comes after some tests and calibration performed with the real robot, in order to find both the "zero" of the joints and their limits.

I have included these limits in the icub-models-generator PR, but they should be also included in the robots-configuration.

CC @pattacini , @traversaro

pattacini, 2021-09-04:
There might be a misunderstanding here.

The changes @AlexAntn did via https://github.com/robotology/icub-models-generator/pull/202 was to align the model's upper roll limit with the physical robot at 163 deg.

#### Physical robot
See https://github.com/robotology/robots-configuration/blob/master/iCubGenova09/hardware/motorControl/left_arm-eb1-j0_1-mc.xml#L14

#### Model
See https://github.com/robotology/icub-models-generator/pull/202#issue-627591424 and in particular:
>small tweak on upper joint limit for the roll to be on par with real robot

---

Thus, I think we can safely close this issue.
cc @traversaro @AlexAntn

pattacini, 2021-09-09:
@traversaro @AlexAntn any comment on this?

traversaro, 2021-09-09:
> 
> 
> @traversaro @AlexAntn any comment on this?

If you check the original discussion that sparked this issue (see https://github.com/robotology/icub-models-generator/pull/202#discussion_r624434849)  the original request was to document this limits somewhere, that was called point 1 , while point 2 (i.e. changing the limits in robots-configuration was not necessary). Point 1 is still not handled, so I am ok closing this issue but if it is ok for you I would open an issue somewhere else to track the documentation of the limits.

pattacini, 2021-09-09:
Yes sure @traversaro 👍🏻 
This issue is pertaining only to point 2 so I'll close it now.

>if it is ok for you I would open an issue somewhere else to track the documentation of the limits.

Agree 👍🏻 
The best place is iCub Tech Documentation, I'd say.

closed 2021-09-09 by pattacini

pattacini, 2021-09-09:
See https://github.com/icub-tech-iit/documentation/issues/174.