robotology/robots-configuration #295: Inconsistencies of joint names

vtikha, 2021-09-28:
We recently noticed that there are some inconsistencies in the joints names of various robots. These inconsistencies are in ~21 robots out of 50. Some have different names, some have underscores some dashes etc...

Here are some examples:

**Head:**

`eyes_version` - > `eyes_vers` (https://github.com/robotology/robots-configuration/search?q=eyes_vers)
`eyes_vergence` -> `eyes_verg` (https://github.com/robotology/robots-configuration/search?q=eyes_verg)

**Right Hand:**

`r_middle_proximal` ->  `r_middle-proximal` (https://github.com/robotology/robots-configuration/search?q=r_middle-proximal)
`r_middle_distal` -> `r_middle-distal` (https://github.com/robotology/robots-configuration/search?q=r_middle-distal)
`r_pinky` -> `r_little-fingers` (https://github.com/robotology/robots-configuration/search?q=r_little-fingers)

This is also valid for the `left hand`. 

We suggest that we modify the joint names using the following standard as reported in the [documentation](https://icub-tech-iit.github.io/documentation/icub_kinematics/icub-joints/icub-joints/)

1. use of full names `eg: eyes_version`
2. use of underscores  `r_little_finger`

randaz81, 2021-09-28:
we probably need to check also if the names of the joints on the icub gazebo models are correct... 
Some time ago I found a similar incontinency on R1 (already fixed)

traversaro, 2021-10-01:
For what regards hands a related issue is https://github.com/robotology/icub-models/issues/30 . For head I need to double check.

PR #296 (2021-10-04): Fix joint names of iCub hands and eyes
Some joint names of the hands and of the eyes have been changed in order to be compliant to the names described in the [documentation](https://icub-tech-iit.github.io/documentation/icub_kinematics/icub-joints/icub-joints/)

It fixes #295
closed 2021-10-04 by pattacini