robotology/robots-configuration #377: Document

traversaro, 2022-08-30:
For people writing new yarprobotinterface files, it would be convenient to document the leves used for `attach`/`detach` phases for different types of devices. By inspecting the configuration files, it seems that these leves are:

| Type of device |  Phase              | Action      | Level           | 
|:--------------:|:-------------------:|:-----------:|:---------------:|
| remapper       | `startup`           | `attach`    |  5              |
| nws            | `startup`           | `attach`    |  10             |
| control/estimation | `startup`       | `attach`    |  15             |
| control/estimation | `shutdown`      | `detach`    |  2              |
| nws            | `shutdown`          | `detach`    |  15             |
| remapper       | `shutdown`          | `detach`    |  20             |

traversaro, 2024-04-12:
I am afraid not. What I think they do (looking into the code) is that they specify the order at which the attach operations are done (the lower the level, the earlier the operation is done). So imagining that the devices are nodes and the "is attached to" or "is detached from" is a ordered arcs, for the attach operations you want to do them in the order of the most closer to the roots of the tree is attached before, and "leafs" are attached later, while for the detach you want to do the contrary.

The relevant code is: https://github.com/robotology/yarp/blob/e4762b0b4a7946bf9379b183379bd32353d54a26/src/libYARP_robotinterface/src/yarp/robotinterface/Robot.cpp#L667-L686 .

pattacini, 2024-04-12:
I can see two different layers of documentation missing here, each describing:
- What an attach/detach level is all about. This probably goes to YARP right away (cc @randaz81). 
- How we generally or typically employ these levels for dealing with robots. We could plan for yielding this documentation at the next PI (cc @Nicogene).

/remind May 06

randaz81, 2024-04-12:
Levels are just numbers to specify a sequence. These numbers do not have any specific meaning. There are some gaps in the sequence to allow the insertion of new stuff without reordering all the sequence. The sequence is dependent on the specific device dependencies n the same you launch modules through yarpmanager. For example first you start a device driver, then you start the network wrapper, finally you do the attach. During closure, the order is obviously  inverted. If you stop the device driver and the wrapper is still running, you'll probaby get a call to a nullptr, resulting in a segfault. I can provide an official documentaion on this topic.