robotology/robots-configuration #378: Unintended change of frequency of iCub's controlBoard_nws_yarp instances from 100 Hz to 50 Hz

traversaro, 2022-08-30:
Historically, iCub's Network Wrapper Servers (i.e. the YARP devices that publish informations over YARP ports) have run at 100 Hz. I may be wrong, but it seems to me that in the latest NWS/NWC layers migration (see https://github.com/robotology/robots-configuration/pull/321 and subsequent PRs), we switched to use 50 Hz frequency. It seems that this happend because we removed all the `<param name="period">10</param>` parameters, meaning that now we rely on the default period of `controlBoard_nws_yarp`, that is 1/0.02 = 50 Hz, see https://github.com/robotology/yarp/blob/v3.7.2/src/devices/ControlBoardWrapper/ControlBoard_nws_ros.h#L41 .

pattacini, 2022-08-31:
Quite an important change that went on unseen, actually.
Thanks @traversaro for the nice catch.

I would ask @randaz81 to comment on this.

For the time being, the quick fix on e.g. iCub 3 could be to make the NWS explicitly run at 100 Hz.
Remember that periods are now in expressed seconds (i.e., 0.01) and no longer milliseconds.

pattacini, 2022-08-31:
We started from iCubGenova01 and spread the changes to the other robots.
Apparently, iCubGenova01 no longer exposes the parameter `period` explicitly.

pattacini, 2022-09-01:
@Nicogene is on it.
The idea is to update the robots' config files to set the period explicitly.

traversaro, 2022-09-01:
Note that the problem was originally spotted by @GiulioRomualdi . 
fyi @S-Dafarra

closed 2022-09-01 by pattacini