robotology/robots-configuration #404: controlboardwrapper2 deprecation

randaz81, 2022-11-29:
device `controlboardwrapper2` was deprecated in previous yarp releases and it will be completely removed in the upcoming yarp 3.8.
The correct devices to use are: `controlBoard_nws_yarp` / `controlboardremapper`.
An update of the robot configuration files is highly recommended, otherwise, the robot will be not able to startup anymore.

pattacini, 2022-11-30:
We ran a [campaign](https://github.com/robotology/robots-configuration/pulls?q=is%3Apr+is%3Aclosed+nws%2Fnwc) in the past to update the robots to the use of the NWS/NWC layers.

However, there are still some [leftovers](https://github.com/robotology/robots-configuration/pulls?q=is%3Apr+is%3Aclosed+nws%2Fnwc), which pertain to:

| Robots/Setups/Folders | Status | Notes |
| :--- | :---: | :--- |
| `iCubLisboa01` | ❌ | Needing fix |
| `R1SN000` | ❌ | Needing fix |
| `iCubTemplates` | ❌ | Needing fix |
| `iRobCub-Mk1` | ❔ | We may have kept it unchanged upon request |
| `experimentalSetups` | ❔ | We need to judge case by case as some setups are old |
| `vizzy` | 👍🏻 | We don't maintain it |

@davidetome who was the main developer in charge may add more insights.

Nicogene, 2023-01-13:
From [this](https://github.com/search?q=repo%3Arobotology%2Frobots-configuration+controlboardwrapper2&type=code) seems that the new `iCubGenova11` is using `controlboardwrapper2`.

Maybe the template for the new robots has to be updated? Otherwise we continue to introduce it

pattacini, 2023-01-13:
>Maybe the template for the new robots has to be updated? Otherwise we continue to introduce it

Very true!
I'll update the OP to include iCubGenov11 too.

cc @AntonioConsilvio @davidetome

davidetome, 2023-01-13:
> From [this](https://github.com/search?q=repo%3Arobotology%2Frobots-configuration+controlboardwrapper2&type=code) seems that the new `iCubGenova11` is using `controlboardwrapper2`.
> 
> Maybe the template for the new robots has to be updated? Otherwise we continue to introduce it

Ciao @Nicogene , actually the templates are not used to create a new robot folder, but only to explain files structure, params meaning and so on.
In fact inside them all robot-files are missing (i.e. https://github.com/robotology/robots-configuration/tree/master/iCubTemplates/iCubTemplateV6_0/hardware/motorControl).
Usually to create a new robot a copy of the last delivery is made.

cc @pattacini @AntonioConsilvio

davidetome, 2023-02-28:
done w/ this PR : 

- https://github.com/robotology/robots-configuration/pull/446

cc @Nicogene @pattacini

closed 2023-02-28 by davidetome