robotology/robots-configuration #405: `grabberDual` deprecation

randaz81, 2022-11-29:
device `grabberDual` was deprecated in previous yarp releases and it will be completely removed in the upcoming yarp 3.8.
The correct device to use is: `frameGrabber_nws_yarp`
An update of the robot configuration files is highly recommended, otherwise, the robot will be not able to startup anymore.
It is also highly recommended to translate the current camera .ini configuration files to the xml format, as shown below [TO BE ADDED]

randaz81, 2022-12-02:
The following configuration, currently used on iCubGenova01 (and similar robots) is bad, because it uses the deprecated device grabberDual and the yarpdev executable, which prevents the integration with ros2.

[image]

randaz81, 2022-12-02:
The following configuration, currently used on iCubGenova09 (and similar robots) is bad, because it uses the yarpdev executable, which prevents the integration with ros2.

[image]

randaz81, 2022-12-02:
EXAMPLE 3:
The following configuration, which uses yarprobotinterface, is the correct one (to be propagated on all robots which use two separated camera device drivers, one for each camera, as shown in EXAMPLE 2)

[image]

randaz81, 2022-12-02:
EXAMPLE 4:
The following configuration, which uses yarprobotinterface, is the correct one (to be propagated on all robots which use a single  camera device driver))

Nicogene, 2023-04-06:
PR in icub-main:

- https://github.com/robotology/icub-main/pull/861