robotology/robots-configuration #462: `inertial` deprecation

Nicogene, 2023-03-28:
Since [YARP 3.8](https://github.com/robotology/yarp/releases/tag/v3.8.0) `inertial` device has been removed in favour of `multipleanalogsensorsserver`

We have to remove all deprecated wrappers xmls https://github.com/search?q=repo%3Arobotology%2Frobots-configuration%20inertial&type=code

traversaro, 2023-03-28:
A bit OT, but I did not know where to write this. One way to catch automatically this kind of regression in the future may be:
* Modify `yarprobotinterface --dryrun` option to also check (or have a separate option for that) if the mentioned device exist, even without opening them (so we can run this check in CI without any robot).
* Add the `yarprobotinterface --dryrun` CI check for each directory of this repo

Note that we can already do the second step, even if it would not have helped in this specific case.

traversaro, 2023-03-28:
By the way, @Nicogene can we rename the issue "`inertial` device removal in YARP 3.8.0" ? "`inertial` deprecation" seems a bit too harmless if you just read the title.

traversaro, 2023-03-28:
Several users were still using the inertial for applications that required to use a port that was publishing something convertible to a `yarp::sig::Vector`, the following issues may be relevant for them:
* https://github.com/robotology/yarp/issues/1693 
* https://github.com/roboticslab-uc3m/teo-configuration-files/issues/27

Nicogene, 2023-04-03:
PR opened: #468

pattacini, 2023-04-04:
Merged!

PR #468 (2023-04-04): Remove inertial xml files and references removed in YARP 3.8.0
It fixes #462 

It will be tested on `iCubGenova11`
closed 2023-04-04 by pattacini