robotology/robots-configuration #470: Replace `analogServer` with `multipleanalogsensorsserver` for publishing FT data

Nicogene, 2023-04-06:
### Task description

As reported by @traversaro in https://github.com/robotology/icub-models/issues/198, some real robots [publish FT data using the old(and deprecated) nws `analogServer`](https://github.com/robotology/robots-configuration/blob/b01aa43c2f51f477cf5b3c855a7e51ec46700ffb/iCubGenova11/icub_all.xml#L97).
This porting to `multipleanalogsensorsserver` has been already done for the simulated robots: https://github.com/robotology/icub-models-generator/pull/231
We should change the xml files in order to align the simulated robot to the real ones and get rid of an nws that will be removed in next releases. 

### Definition of done

The changes have been made, the PR has been merged

cc @traversaro @pattacini @gabrielenava

pattacini, 2023-04-15:
/remind next Monday

Nicogene, 2023-04-18:
PR opened: https://github.com/robotology/robots-configuration/pull/482

To be tested on `iCubGenova11`

Nicogene, 2023-04-20:
I noticed that `wholebodydynamics` in icub-main have never been ported to the usage of MAS server/client:

https://github.com/robotology/icub-main/blob/ec6123471ee3cf212ef45fdda772e2453ed97208/src/modules/wholeBodyDynamics/observerThread.cpp#L411-L433

So maybe we need to adapt it before doing this porting

Nicogene, 2023-04-20:
PR merged ✔️

commit 2b7175ef (2023-04-19): FT: replace analogServer (deprecated) with multipleanalogsensorsserver

It fixes #470
commit bfca665d (2023-04-19): FT: replace analogServer (deprecated) with multipleanalogsensorsserver

It fixes #470
commit 08376922 (2023-04-19): FT: replace analogServer (deprecated) with multipleanalogsensorsserver

It fixes #470
commit 91b50021 (2023-04-19): FT: replace analogServer (deprecated) with multipleanalogsensorsserver

It fixes #470