robotology/robots-configuration #479: FIX Dry-Run errors

pattacini, 2023-04-14:
Following up on #466, we're required to start tackling the errors spotted via the CI Dry-Run.

Check out, for example, https://github.com/robotology/robots-configuration/actions/runs/4699186971.

cc @Nicogene

pattacini, 2023-04-17:
```console
75% tests passed, 38 tests failed out of 150

Total Test time (real) = 311.01 sec

The following tests FAILED:
	  3 - dry-run_R1SN002_xml (Failed)
	 13 - dry-run_iCubAberystwyth01_xml (Failed)
	 15 - dry-run_iCubAbuDhabi01_xml (Failed)
	 17 - dry-run_iCubAnkara01_xml (Failed)
	 21 - dry-run_iCubBielefeld01_xml (Failed)
	 23 - dry-run_iCubBielefeld02_xml (Failed)
	 25 - dry-run_iCubBielefeld03_xml (Failed)
	 41 - dry-run_iCubFrankfurt01_xml (Failed)
	 49 - dry-run_iCubGenova04_xml (Failed)
	 50 - dry-run_iCubGenova04_dev (Failed)
	 58 - dry-run_iCubGenova09_dev (Failed)
	 63 - dry-run_iCubGrenoble01_xml (Failed)
	 67 - dry-run_iCubHeidelberg01_xml (Failed)
	 68 - dry-run_iCubHeidelberg01_dev (Failed)
	 69 - dry-run_iCubHertfordshire01_xml (Failed)
	 71 - dry-run_iCubHertfordshire02_xml (Failed)
	 75 - dry-run_iCubKarlsruhe01_xml (Failed)
	 77 - dry-run_iCubLausanne01_xml (Failed)
	 81 - dry-run_iCubLethbridge01_xml (Failed)
	 83 - dry-run_iCubLisboa01_xml (Failed)
	 84 - dry-run_iCubLisboa01_dev (Failed)
	 85 - dry-run_iCubLisboa02_xml (Failed)
	 94 - dry-run_iCubMoscow01_dev (Failed)
	 95 - dry-run_iCubMunich01_xml (Failed)
	101 - dry-run_iCubOffenbach01_xml (Failed)
	103 - dry-run_iCubOffenbach02_xml (Failed)
	105 - dry-run_iCubParis01_xml (Failed)
	107 - dry-run_iCubParis02_xml (Failed)
	109 - dry-run_iCubPisa01_xml (Failed)
	111 - dry-run_iCubPlymouth01_xml (Failed)
	113 - dry-run_iCubPlymouth02_xml (Failed)
	117 - dry-run_iCubPrototype01_xml (Failed)
Errors while running CTest
	119 - dry-run_iCubRome02_xml (Failed)
	124 - dry-run_iCubShanghai01_dev (Failed)
	131 - dry-run_iCubTokyo01_xml (Failed)
	135 - dry-run_iCubUrbana01_xml (Failed)
	143 - dry-run_iCubZurich01_xml (Failed)
	149 - dry-run_vizzy_xml (Failed)
```

pattacini, 2023-04-17:
Solving #477 allowed us to sort out `iCubGenova09` but not yet `iCubGenova04`.
See https://github.com/robotology/robots-configuration/actions/runs/4720445946.

martinaxgloria, 2023-04-18:
Today I started investigating the log files relative to the test that failed (@pattacini I'm not able to move the issue in progress). I observed that some robots gave this error:
```
[DEBUG] Reading file /dependencies/install/share/ICUBcontrib/robots/iCubAberystwyth01/./icub_all.xml
[WARNING] Invalid syntax while loading /dependencies/install/share/ICUBcontrib/robots/iCubAberystwyth01/./icub_all.xml at line 1 . No DTD found. Assuming version yarprobotinterfaceV3.0
[ERROR] DTD V1.x has been deprecated. Please update your configuration files to DTD v3.x
[ERROR] Invalid DTD version, execution stopped.
```

Opening the one of the files that caused the error, I saw that `icub_all.xml` hasn't the code line:
```xml
<!DOCTYPE robot PUBLIC "-//YARP//DTD robotInterface 3.0//EN" "http://www.yarp.it/DTD/robotInterfaceV3.0.dtd">
```

By adding this line, is possible to fix the error? @pattacini

pattacini, 2023-04-18:
>I'm not able to move the issue in progress

This is ok because you don't have `write` access to this repo (as required by ZenHub to operate). I'll move it myself.

>By adding this line, is possible to fix the error?

Yep, it should suffice.

martinaxgloria, 2023-04-18:
> Yep, it should suffice.

After adding that line, some errors were solved:

[image]

I will investigate the others.

martinaxgloria, 2023-04-18:
After chatting with @Nicogene, we found out that the target was right except for a typo. I fixed it and now the test for those robots passes successfully.

martinaxgloria, 2023-04-19:
Moreover, there are 10 robots that return this failure:
```
[ERROR] |yarp.os.Property| cannot read from yarprobotinterface.ini
[FATAL] Missing "config" argument
```
Analyzing them, I saw that they don't have the `yarprobotinterface.ini` file and they miss also the .xml file to pass as `config` argument. Is it possible to skip the dry-run for them?

cc @Nicogene @pattacini

pattacini, 2023-04-19:
> Is it possible to skip the dry-run for them?

Nope 😉 
We need to make up the easiest `yarprobointerface.ini` possible.

martinaxgloria, 2023-04-19:
`iCubGenova04` gives out this error:
```
[ERROR] Target device left_hand_mc (network = left_hand_joints ) does not exist.
[ERROR] Cannot run attach action on device alljoints-mc_remapper
```
In fact, the `alljoints-mc_remapper.xml` is: 

```xml
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE devices PUBLIC "-//YARP//DTD yarprobotinterface 3.0//EN" "http://www.yarp.it/DTD/yarprobotinterfaceV3.0.dtd">

<device xmlns:xi="http://www.w3.org/2001/XInclude" name="alljoints-mc_remapper" type="controlboardremapper">
    <paramlist name="networks">
        <elem name="head_joints">       (  0   5   0  5 )</elem>
        <elem name="face_joints">       (  0   1   0  1 )</elem>
        <elem name="left_arm_joints">   (  6  12   0  6 )</elem>
        <elem name="left_hand_joints">  ( 13  21   0  8 )</elem>
        <elem name="right_arm_joints">  ( 22  28   0  6 )</elem>
        <elem name="right_hand_joints"> ( 29  37   0  8 )</elem>
        <elem name="torso_joints">      ( 38  40   0  2 )</elem>
        <elem name="left_leg_joints">   ( 41  46   0  5 )</elem>
        <elem name="right_leg_joints">  ( 47  52   0  5 )</elem>
    </paramlist>
    <param name="joints"> 53 </param>
    <action phase="startup" level="5" type="attach">
        <paramlist name="networks">
            <elem name="head_joints">       head-mc_remapper </elem>
            <elem name="face_joints">       face_mc </elem>
            <elem name="left_arm_joints">   left_arm-mc_remapper </elem>
            <elem name="left_hand_joints">  left_hand_mc </elem>
            <elem name="right_arm_joints">  right_arm-mc_remapper </elem>
            <elem name="right_hand_joints"> right_hand_mc </elem>
            <elem name="torso_joints">      torso-mc_remapper </elem>
            <elem name="left_leg_joints">   left_leg-mc_remapper </elem>
            <elem name="right_leg_joints">  right_leg-mc_remapper </elem>
        </paramlist>
    </action>
    <action phase="shutdown" level="20" type="detach" />
</device>

```
There are no file relative to the elem name `face_joints`, `left_hand_joints` and `right_hand_joints` in `iCubGenova04/wrappers/motorControl`. How can I proceed?

cc @pattacini @Nicogene

martinaxgloria, 2023-04-19:
About R1SN002, instead, the error regards the `rpLidar wrappers` which are of type `Rangefinder2DWrapper`. I found [here](https://github.com/robotology/yarp/pull/2827) that it is deprecated in favor of `rangefinder2D_nws_yarp` and `rangefinder2D_nws_ros`. Can I remove the deprecated device and substitute it with the supported one?

pattacini, 2023-04-19:
>About R1SN002, instead, the error regards the rpLidar wrappers which are of type Rangefinder2DWrapper. I found https://github.com/robotology/yarp/pull/2827 that it is deprecated in favor of rangefinder2D_nws_yarp and rangefinder2D_nws_ros. Can I remove the deprecated device and substitute it with the supported one?

Yep, get inspired by what's done for R1SN003.

martinaxgloria, 2023-04-19:
Today, @Nicogene helps me in fixing some of the dry-run errors and now the ctest gives this result:

[image]

I will investigate the last ones.

pattacini, 2023-04-20:
Fixed test 94: `dry-run_iCubMoscow01_dev` on robotology@master via 134864b20f7f1d0f86ed01d57c8cfd485e0d17ef.
@martinaxgloria, you may consider rebasing your fork@master on top of it.

Nicogene, 2023-04-20:
I would check if #482 generated new errors 😬

pattacini, 2023-04-20:
> I would check if #482 generated new errors 😬

Dry-Run in https://github.com/robotology/robots-configuration/actions/runs/4753025058.

pattacini, 2023-04-20:
> > I would check if #482 generated new errors 😬
> 
> Dry-Run in https://github.com/robotology/robots-configuration/actions/runs/4753025058.

```console
77% tests passed, 35 tests failed out of 150

Total Test time (real) = 301.69 sec

The following tests FAILED:
	  3 - dry-run_R1SN002_xml (Failed)
	 13 - dry-run_iCubAberystwyth01_xml (Failed)
	 15 - dry-run_iCubAbuDhabi01_xml (Failed)
	 17 - dry-run_iCubAnkara01_xml (Failed)
	 21 - dry-run_iCubBielefeld01_xml (Failed)
	 23 - dry-run_iCubBielefeld02_xml (Failed)
	 25 - dry-run_iCubBielefeld03_xml (Failed)
	 41 - dry-run_iCubFrankfurt01_xml (Failed)
	 49 - dry-run_iCubGenova04_xml (Failed)
	 50 - dry-run_iCubGenova04_dev (Failed)
	 63 - dry-run_iCubGrenoble01_xml (Failed)
	 67 - dry-run_iCubHeidelberg01_xml (Failed)
	 68 - dry-run_iCubHeidelberg01_dev (Failed)
	 69 - dry-run_iCubHertfordshire01_xml (Failed)
	 71 - dry-run_iCubHertfordshire02_xml (Failed)
	 75 - dry-run_iCubKarlsruhe01_xml (Failed)
	 77 - dry-run_iCubLausanne01_xml (Failed)
	 81 - dry-run_iCubLethbridge01_xml (Failed)
	 83 - dry-run_iCubLisboa01_xml (Failed)
	 84 - dry-run_iCubLisboa01_dev (Failed)
	 85 - dry-run_iCubLisboa02_xml (Failed)
	 95 - dry-run_iCubMunich01_xml (Failed)
	101 - dry-run_iCubOffenbach01_xml (Failed)
	103 - dry-run_iCubOffenbach02_xml (Failed)
	105 - dry-run_iCubParis01_xml (Failed)
	107 - dry-run_iCubParis02_xml (Failed)
	109 - dry-run_iCubPisa01_xml (Failed)
	111 - dry-run_iCubPlymouth01_xml (Failed)
	113 - dry-run_iCubPlymouth02_xml (Failed)
	117 - dry-run_iCubPrototype01_xml (Failed)
	119 - dry-run_iCubRome02_xml (Failed)
	131 - dry-run_iCubTokyo01_xml (Failed)
	135 - dry-run_iCubUrbana01_xml (Failed)
	143 - dry-run_iCubZurich01_xml (Failed)
	149 - dry-run_vizzy_xml (Failed)
```

pattacini, 2023-04-20:
I'm afraid that #482 will cause lots of conflicts while merging @martinaxgloria's work...

pattacini, 2023-04-20:
@martinaxgloria can you open up a PR against `robotology@master` for what you've done so far?
Too big PR's are difficult to handle.

Then, we will proceed incrementally.

martinaxgloria, 2023-04-20:
PR:

- https://github.com/robotology/robots-configuration/pull/486

pattacini, 2023-04-20:
Merged, thanks!

pattacini, 2023-04-20:
I managed to merge `msster` into `devel` with no conflicts at all 🎉 
@martinaxgloria, you can start over with forking from `robotology@master`.

pattacini, 2023-04-20:
### Robots still failing in `robotology@master`:
```console
 21 - dry-run_iCubBielefeld01_xml (Failed)
 23 - dry-run_iCubBielefeld02_xml (Failed)
 25 - dry-run_iCubBielefeld03_xml (Failed)
 41 - dry-run_iCubFrankfurt01_xml (Failed)
 75 - dry-run_iCubKarlsruhe01_xml (Failed)
 85 - dry-run_iCubLisboa02_xml (Failed)
101 - dry-run_iCubOffenbach01_xml (Failed)
103 - dry-run_iCubOffenbach02_xml (Failed)
109 - dry-run_iCubPisa01_xml (Failed)
117 - dry-run_iCubPrototype01_xml (Failed)
135 - dry-run_iCubUrbana01_xml (Failed)
143 - dry-run_iCubZurich01_xml (Failed)
```

### Robots still failing in `robotology@devel`:
```console
 12 - dry-run_ergoCubSN000_dev (Failed)
 17 - dry-run_iCubAnkara01_xml (Failed)
 19 - dry-run_iCubBarcelona01_xml (Failed)
 21 - dry-run_iCubBielefeld01_xml (Failed)
 23 - dry-run_iCubBielefeld02_xml (Failed)
 25 - dry-run_iCubBielefeld03_xml (Failed)
 41 - dry-run_iCubFrankfurt01_xml (Failed)
 75 - dry-run_iCubKarlsruhe01_xml (Failed)
 85 - dry-run_iCubLisboa02_xml (Failed)
101 - dry-run_iCubOffenbach01_xml (Failed)
103 - dry-run_iCubOffenbach02_xml (Failed)
109 - dry-run_iCubPisa01_xml (Failed)
117 - dry-run_iCubPrototype01_xml (Failed)
135 - dry-run_iCubUrbana01_xml (Failed)
143 - dry-run_iCubZurich01_xml (Failed)
```

meaning that #482 did introduce regressions on the following robots:
- `ergoCubSN000`
- `iCubAnkara01`
- `iCubBarcelona01` 

@Nicogene can you have a look at those with the help of @martinaxgloria?

pattacini, 2023-04-20:
Decided in https://github.com/robotology/robots-configuration/commit/af8bc53a4045040cbce18077731e80eb1bde775b to be milder for those robots w/o `yri.ini`.

Here's the test outcome for `robotology@master`:
```console
109 - dry-run_iCubPisa01_xml (Failed)
```

Just one robot left, @martinaxgloria! 🎉

pattacini, 2023-04-20:
...plus the 3 robots in devel.

pattacini, 2023-04-20:
> > * `ergoCubSN000`
> > * `iCubAnkara01`
> > * `iCubBarcelona01`
> 
> @Nicogene and I tried to run the dry-run for those robots from `devel` but the test passed successfully. This is the output: [image]
> 
> which are the robots that fail in `robotology@master` as well.

Let me try again then.

pattacini, 2023-04-20:
So, after a clean restart, robots failing in `robotology@devel` are:
```console
 49 - dry-run_iCubGenova04_xml (Failed)
 50 - dry-run_iCubGenova04_dev (Failed)
109 - dry-run_iCubPisa01_xml (Failed)
```

Here's below the problem w/ `iCubGenov04`:
```console
[ERROR] Failed to open file  file /dependencies/install/share/ICUBcontrib/robots/iCubGenova04/./wrappers/FT/right_foot-FT_wrapper_multipleSens.xml included by  icub_wbd.xml at line 96
```

cc @martinaxgloria @Nicogene

Nicogene, 2023-04-20:
> So, after a clean restart, robots failing in `robotology@devel` are:
> 
> ```
>  49 - dry-run_iCubGenova04_xml (Failed)
>  50 - dry-run_iCubGenova04_dev (Failed)
> 109 - dry-run_iCubPisa01_xml (Failed)
> ```
> 
> Here's below the problem w/ `iCubGenov04`:
> 
> ```
> [ERROR] Failed to open file  file /dependencies/install/share/ICUBcontrib/robots/iCubGenova04/./wrappers/FT/right_foot-FT_wrapper_multipleSens.xml included by  icub_wbd.xml at line 96
> ```
> 
> cc @martinaxgloria @Nicogene

I forgot to save and then commit some files of `iCubGenova04`, now should be fixed: https://github.com/robotology/robots-configuration/commit/bfbbe7d13a8a8dbaaf5f47586d1ae5fd4522ecf8

cc @pattacini

pattacini, 2023-04-20:
Fantastic! I confirm that now `devel` is fine.

Here's below the log of the very last robot failing in `master` (`dry-run_iCubPisa01_xml`):
```console
[ERROR] Target device torso-mc_remapper (network = torso ) does not exist.
[ERROR] Cannot run attach action on device left_arm_cartesian
[ERROR] Target device torso-mc_remapper (network = torso ) does not exist.
[ERROR] Cannot run attach action on device right_arm_cartesian 
```

cc @martinaxgloria