robotology/robots-configuration #485: Replace analogServer with multipleanalogsensorsserver for publishing FT data – Stint 2

Nicogene, 2023-04-20:
This is the followup of #470 

We would like to:

> One thing that we may took the occasion to do know that we do a breaking change is that we could group in a single wrapper all the FT sensors of the leg, for consistency with what we do for other wrappers, such as controlboard.

Nicogene, 2023-05-17:
The PR #505 has been merged, I will then proceed to apply those changes to all the robots.

cc @pattacini @traversaro

Nicogene, 2023-05-17:
PR: #517 

I will test it on `iCubGenova11` tomorrow. I am not sure it will be cover all the cases, I hope the dry-run will help in this sense

pattacini, 2023-05-17:
>I am not sure it will be cover all the cases, I hope the dry-run will help in this sense

🤞🏻

Nicogene, 2023-05-18:
Today @traversaro and I discovered that after #482 all the robots that use `embObjStrain` have the publishing of the FT broken because `embObjStrain` has not been ported to `IMultipleAnalogSensorsInterfaces` (see https://github.com/robotology/icub-main/blob/001b897fb29aaa1cb58ff713361fc68b2f97e1ca/src/libraries/icubmod/embObjStrain/embObjStrain.h#L52-L55)

This is not caught by the CI because right now `multipleanalogsensorsserver` does not fail if the device attached does not implement any interface (see https://github.com/robotology/yarp/blob/872679fa605b95163da4a64e61c161ab9df3d485/src/devices/multipleanalogsensorsserver/MultipleAnalogSensorsServer.cpp#L403-L413)
In this sense, we can add that if all view fail we throw an error but this will not solve the problem will only make the ci fail for the robots we broke.

We spoke w/ @marcoaccame and we checked that right now both `embObjFTSensor` and `embObjMultipleFTSensor` (the ones compatible w/ MAS pipeline) does not support right now `strain` board.
We would need a change `embobjFTSensor` in `icub-main` for ignoring the temperature/imu sensor when using `strain`, on fw side we do not need anything. In this way, we could throw away `embObjStrain` since it is used only w/ `analogServer`.
Moreover, we need to migrate the xml files of strain to the "new" device.

The last resort is to roll back the robot that is now broken to `analogServer` but it would be a pity.

cc @pattacini @valegagge

pattacini, 2023-05-18:
Thanks for the summary 👍🏻 

We will be discussing this the next Tue at the board meeting.

Nicogene, 2023-05-18:
Putting on hold since we need to takle this: https://github.com/robotology/icub-main/issues/873

pattacini, 2023-05-25:
> Putting on hold since we need to takle this: [robotology/icub-main#873](https://github.com/robotology/icub-main/issues/873)

Done!

Nicogene, 2023-05-26:
We can finally close this issue, we will track the work on the simulators on a different one.

PR #517 (2023-05-25): Migrate iCub FT to have one NWS for each part
This is the follow-up of #505.

It fixes #485 

TO BE TESTED ON `iCubGenova02`
closed 2023-05-26 by Nicogene