robotology/robots-configuration #509: CI failing on `ergoCubSN000` and `iCubGenova09` with the latest `yarp-3.8`

pattacini, 2023-05-11:
I've just regenerated the docker image used by the CI relying on the `yarp-3.8` branch rather than the `v3.8.0` release (#508) and I'm getting the following error:

[dry-run_log_ergoCubSN000_dev.txt](https://github.com/robotology/robots-configuration/files/11454325/dry-run_log_ergoCubSN000_dev.txt):
```console
[ERROR] embObjMultipleFTsensors: BOARD left_arm-eb2-j0_1 (IP 10.0.1.2)   getTemperatureSensorMeasure() fails data for index: 0  not found
```

[dry-run_log_iCubGenova09_dev.txt](https://github.com/robotology/robots-configuration/files/11454356/dry-run_log_iCubGenova09_dev.txt):
```console
[ERROR] embObjMultipleFTsensors: BOARD left_leg-eb8-j3_5 (IP 10.0.1.8)   getTemperatureSensorMeasure() fails data for index: 0  not found
```

@traversaro, will https://github.com/robotology/yarp/pull/2967 fic this problem too?

traversaro, 2023-05-11:
This is more of an issue of `embObjMultipleFTsensors`, but indeed it will be mitigated by https://github.com/robotology/yarp/pull/2967 .

It is related to the second fix from https://github.com/robotology/icub-main/pull/871, and actually the reason why now it appears is that we fixed that bug. Before then, the value returned by `embObjMultipleFTsensors::getNrOfTemperatureSensors` was zero if no temperature measurement was received, and so no one called `embObjMultipleFTsensors::getTemperatureSensorMeasure` . https://github.com/robotology/yarp/pull/2967 ensures that `embObjMultipleFTsensors::getTemperatureSensorMeasure` is not called if the status of the sensors is not `MAS_OK`, so it should ensure that this error does not appear.

pattacini, 2023-05-11:
Thanks for the clarification!

Does your last doubt mean that the driver per se in this configuration is not ready to be dry run at all?

traversaro, 2023-05-11:
It is not the driver, it is the `multipleanalogsensorsserver` that will print errors if no valid sensor measure is available. I wonder how other network wrapper server deal with this, perhaps they do not complain and just do not publish data if it is not available?

pattacini, 2023-05-11:
Yep, with the term `driver` I was referring to the whole stack.

This can be a major issue for the CI. In the worst case, I may try to disable these layers during the CI as done for others, but it'd be a pity after all.

We can discuss it.
Thanks

traversaro, 2023-05-11:
I think it could make sense if we agree on a `dryrun` or similar parameter to pass to every device so that every device is aware and suppress errors that do not make sense to print during a dryrun (such as errors because no sensor message arrived).

pattacini, 2023-05-12:
Temporary disabled the failing tests in the CI via https://github.com/robotology/robots-configuration/commit/93d60a4bab2f7e6d0c7f1fa1876bbb05c4caaa7d.

Failing tests can be still run by:
```console
ctest -L failing
```

cc @traversaro

pattacini, 2023-05-12:
https://github.com/robotology/robots-configuration/commit/ea682368199ab3978bc71a85f9c0febb8c5319dc streamlines the handling of skipped tests.

Now, to run locally or on Gitpod skipped tests, one needs to do:
```console
ctest -L skip
```

Tests to be skipped are specified in:
- https://github.com/robotology/robots-configuration/blob/master/tests/skip_tests.txt

pattacini, 2023-05-12:
Aligend T2T w/ @traversaro: all good then 👍🏻 
The error message is not causing any harm to the CI.

The CI failed because of the yarprobotinterface crash, which will be solved via https://github.com/robotology/yarp/pull/2967.

Putting in review.

pattacini, 2023-05-15:
The PR https://github.com/robotology/yarp/pull/2967 has been merged although the dry-run still fails with the following log:
```console
[ERROR] yarprobotinterface intercepted a segmentation fault caused by a faulty plugin:
[ERROR] /dependencies/install/lib/yarp/yarp_multipleanalogsensorsserver.so(+0x13f69) [0x7f42b401cf69]
```

@traversaro is on it.

traversaro, 2023-05-15:
> @traversaro is on it.

My bad, apparently I forgot to push some changes, the fix is in https://github.com/robotology/yarp/pull/2970 .

commit 17107ceb (2023-05-16): fix #509
closed 2023-05-16 by pattacini