robotology/robots-configuration #545: [ergoCubSN000] Gear ratio values of right and left shoulder roll joints do not match

isorrentino, 2023-07-20:
Hi all, while checking the parameters needed to control the motors in torque, I noticed that the gear ratio values of the right and left shoulder roll joints do not match. Do they have a different harmonic drive or is this an error? (See the parameter `Gearbox_M2J`).

- right
  https://github.com/robotology/robots-configuration/blob/c7c965a9aa1b8295ace3a32c2fb69c18c18794db/ergoCubSN000/hardware/mechanicals/right_arm-eb1-j0_1-mec.xml#L5-L19

- left
  https://github.com/robotology/robots-configuration/blob/c7c965a9aa1b8295ace3a32c2fb69c18c18794db/ergoCubSN000/hardware/mechanicals/left_arm-eb2-j0_1-mec.xml#L5-L19

cc @lrapetti @ale-git

pattacini, 2023-07-20:
Thanks for reporting.

It's most likely an error, but I'm checking that out w/ @Mick3Lozzo.
Once done, I'll apply a patch.

Mick3Lozzo, 2023-07-20:
[image]

gear ratio = 100 @isorrentino 

cc @pattacini

PR #546 (2023-07-20): fixed gear ratio right shoulder roll
Fix #545.
closed 2023-07-20 by pattacini