robotology/robots-configuration #569: Check and validate ergoCub wrist-mk2.x gear ratios

pattacini, 2023-09-19:
As of now, the FW relies on these values:

https://github.com/robotology/robots-configuration/blob/151cd3936d34d5d2a59f0c2de66ecb3a6e27d2bd/ergoCubSN000/hardware/mechanicals/left_arm-eb31-j4_6-mec.xml#L14-L15

We need to check their correctness (both the ratios) wrt the CAD and the fact that the mechanism is coupled.

Here's our present documentation saying that the motor speed-reducer ratio is `196:1` and the belt drive ratio is `2:1`.

Perhaps, the parallel kinematics has an impact here, in the way we'd need to convert those values in the conf files @ale-git?

cc @valegagge @lawproto @mfussi66 @Nicogene @andreaderito

pattacini, 2023-09-19:
Weird enough, $`196 \cdot 1.778 = 348.44`$, which is similar to 384.44 but with 2 numbers swapped.

mfussi66, 2023-09-19:
> and the belt drive ratio is `2:1`

This is caused by the number of teeth of the two pulleys: 28/14.

Lawproto, 2023-09-19:
> Weird enough, 196×1.778=348.44, which is similar to 384.44 with 2 digits swapped.

Yeah, @pattacini! @simeonedussoni noticed the same thing. And, weird enough again, the number with the digits swapped is close enough to `196x2=396`, which is the number I would expect given our current parameters. It seems that two wrongs *do* make a right, sometimes.

> This is caused by the number of teeth of the two pulleys: 28/14.
Given the `1.778` oddball, I checked on the teeth of the pulleys again. I confirm it is as @mfussi66 said.

[image]
[image]

I have no idea where that number came from in the first place.

pattacini, 2023-09-19:
PR opened in:
- #570 

@simeonedussoni @valegagge, remember to update the values in the test setup of the new wrist.

mfussi66, 2023-09-19:
The E2J parameter inevitably has an effect on the decoupler, since it uses the angle of the driven pulley to compute the direct kinematics.

closed 2023-09-19 by pattacini

maggia80, 2023-09-20:
looking at the model in Simscape it turns out that: [image]

the ratio is 1.777

pattacini, 2023-09-20:
I'll check it out and, if it's the case, I'll open an issue at https://github.com/robotology/icub-firmware-models.

pattacini, 2023-09-20:
Hi @maggia80 

We do generate the code of the controller only as per https://github.com/robotology/icub-firmware-models/tree/master/wrist_mk2_1#embedded-firmware-generation.

The ratio is hand-coded in the model but used only for simulation.
At any rate, I'll open up a PR to fix them up.

pattacini, 2023-09-20:
Here's the PR:
- https://github.com/robotology/icub-firmware-models/pull/57

Lawproto, 2023-09-20:
I would like to take a look at the Simscape model, where can I find it?

pattacini, 2023-09-20:
> I would like to take a look at the Simscape model, where can I find it?

You ought to download the repo https://github.com/robotology/icub-firmware-models and then open up the MATLAB project https://github.com/robotology/icub-firmware-models/blob/master/wrist_mk2_1/spherical_wrist.prj (using the latest release R2023b).

PR #570 (2023-09-19): Fix ergoCub wrist gear ratios
Fix #569.

$`392 = 196 \times 2`$.