robotology/robots-configuration #582: Wrong eyes joint names for iCubGenova11

xEnVrE, 2023-10-31:
For some reason, the name of the eyes version and vergence for this robot are `eyes_vers` and `eyes_verg` 

https://github.com/robotology/robots-configuration/blob/0891fbec0448283c5f62fe91f6b7c4e98b4cac8e/iCubGenova11/hardware/mechanicals/head-eb21-j2_5-mec.xml#L12

which are different from:
- the documentation https://icub-tech-iit.github.io/documentation/icub_kinematics/icub-joints/icub-joints/#head-20
- the simulation https://github.com/robotology/icub-models/blob/b2391b90432e13f053a2b516e2d423a94f3e84e1/iCub_manual/conf_manual/iCubGazeboV2_5_visuomanip/gazebo_icub_eyes.ini#L15
- those of other robots, e.g., https://github.com/robotology/robots-configuration/blob/0891fbec0448283c5f62fe91f6b7c4e98b4cac8e/iCubGenova10/hardware/mechanicals/head-eb21-j2_5-mec.xml#L12

This is creating issues when trying to use the same software in simulation and on the real setup as joints names are different.

Can I open a PR to fix that?

cc @pattacini

sgiraz, 2023-10-31:
Hi @xEnVrE,

It seems that the typo exists from the beginning (see https://github.com/robotology/robots-configuration/pull/399).

pattacini, 2023-10-31:
> Can I open a PR to fix that?

Sure!

Would you be willing to fix the following robots as well?
- iCubValparaiso01
- iCubLisboa01
- iCubShanghai01

This came out from a search: https://github.com/search?q=repo%3Arobotology%2Frobots-configuration+eyes_verg&type=code&p=1.

The PR should be against `master` as it's a fix.

xEnVrE, 2023-10-31:
> Would you be willing to fix the following robots as well?

Done!

closed 2023-10-31 by pattacini

pattacini, 2023-10-31:
Thank you @xEnVrE!

PR #583 (2023-10-31): Fix wrong eyes joints names of several robots
As per the title.

Fixes #582