robotology/robots-configuration #606: Port all the robots' configuration to the TemperatureSensorType parameter

Nicogene, 2024-01-08:
@AntonioConsilvio pointed me out that this PR:
- https://github.com/robotology/icub-main/pull/911
Remove the old parameter `HasTempSensor` replacing it with `TemperatureSensorType`, this means that on icub-main@devel several robots cannot start right now, see:

- https://github.com/search?q=repo%3Arobotology%2Frobots-configuration%20HasTempSensor&type=code

cc @valegagge @MSECode

MSECode, 2024-01-08:
Since the code is retro-compatible, it should start even if using the old parameter `HasTempSensor` if this one is set to zero. As shown by the tests done here:
https://github.com/icub-tech-iit/five/issues/178#issuecomment-1785322184

valegagge, 2024-01-08:
> Since the code is retro-compatible, it should start even if using the old parameter `HasTempSensor` if this one is set to zero. As shown by the tests done here: [icub-tech-iit/five#178 (comment)](https://github.com/icub-tech-iit/five/issues/178#issuecomment-1785322184) So if we do not have the type of sensor on the motor or do we not need it, we can leave the configurations as they are But i'll check it

it is correct! we don't need to update the configuration of all robots unless the user want to read the temperature sensor in case the robot has it.

So I think we can close this issue

closed 2024-01-09 by Nicogene