robotology/robots-configuration #613: Register iCub "iCubManchester01" and update robotology-superbuild

albertolillo23, 2024-02-13:
Hi,
we are based in University of Manchester and we have an iCub in our lab with a version of robotology-superbuild 1.2.
We want to update the robotology but the name is not in the page of list of iCub.

Name: iCubManchester01

Could you please help us to register iCub and obtain the correct configuration file?

Thank you,

CoRoLab University of Manchester

pattacini, 2024-02-13:
Hi @albertolillo23 

Thanks for reaching out.
We will evaluate this request in terms of the required activities and get back to you with an estimate.

cc @Nicogene

albertolillo23, 2024-02-13:
For more information, I add that the version of yarp is 2.3.72.1
Thanks

Nicogene, 2024-02-22:
Hi @albertolillo23!

From:
- https://github.com/robotology/robots-configuration/pull/533

We decided to put the configuration files of the robots we didn't have news of for a while in the `robot-icebox` folder to shorten the list of maintained robots and the required efforts for keeping this repo up-to-date.

Having YARP `2.3.72` means that on your robot the software you are running on the robot is stuck to ~ 2018, maybe for you is sufficient to upgrade it to just **before** the institution of robot-icebox, i.e. the [Distro 2023.05.0](https://github.com/orgs/robotology/discussions/639)?

This should be possible right now without the need of any intervention.

For upgrading, I suggest starting from a clean state, and follow these instructions
- for the workstation -> https://icub-tech-iit.github.io/documentation/sw_installation/linux_from_sources_superbuild/ 
- for the `pc104` -> https://icub-tech-iit.github.io/documentation/sw_installation/icub_head_superbuild/

using the [ `v2023.05.0`](https://icub-tech-iit.github.io/documentation/sw_versioning_table/2023.05.0/) tag.

Please let us know if it is sufficient, if you instead want to put your robot out from the `robot-icebox` more actions are needed, first of all, changes to the configuration files are needed to make them compatible with the latest software changes.
A robot similar to yours that can be taken as example is `iCubLondon01` but it took also several hw maintenance to make it up-to-date:

- https://github.com/robotology/icub-tech-support/issues/1171

Nicogene, 2024-03-06:
Hi @albertolillo23!

any update on this? Have you managed to update to the distro 2023.05.0?

albertolillo23, 2024-03-06:
Hi @Nicogene!

Sorry for the late reply, two days ago we tried to do the update. 

Following the guide in the link regarding `pc104`, I noticed that in our pc, inside the robot folder we do not have a pre-existing "robotology-superbuild" folder, but I only have these folders:

- icub-basic-demos
- iCubContrib
- icub-contrib-common
- icub-firmware-build
- icub-firmware-shared
- icub-main
- robots-configuration
- yarp

So I downloaded it and followed the compile commands. 

When I run the command (in the path: /usr/local/src/robot/robotology-superbuild):

- `cmake -DROBOTOLOGY_USES_GAZEBO:BOOL=OFF -DROBOTOLOGY_ENABLE_ICUB_HEAD:BOOL=ON ..` 

I get an error saying:

- "CMake Error: The source directory "/usr/local/src/robot" does not appear to contain CMakeLists.txt".

How can we solve it? 

Thanks for the support.

Nicogene, 2024-03-06:
Hi @albertolillo23 you need to do the following steps:

```
cd robotology-superbuild
mkdir build && cd build
cmake -DROBOTOLOGY_USES_GAZEBO:BOOL=OFF -DROBOTOLOGY_ENABLE_ICUB_HEAD:BOOL=ON ..
make
```
Once the build is finished you need to add this line at the beginning of the bashrc

```
source /usr/local/src/robot/robotology-superbuild/build/install/share/robotology-superbuild/setup.sh
```

pattacini, 2024-03-07:
>Once the build is finished you need to add this line at the beginning of the bashrc

Probably, it could also help cleaning up the content of bashrc by removing any reference to `YARP_*` and `ICUB_*` env variables, including relevant paths appended to `PATH` and `LD_LIBRARY_PATH`.

Nicogene, 2024-03-20:
Hi @albertolillo23!

Did you manage to install the distro 2023.05.0 via robotology-superbuild?

albertolillo23, 2024-03-20:
Hi @Nicogene,
sorry for the slow response, but we don't always have the availability of the lab where iCub is located, we will try it on Thursday and as soon as it is done we will let you know.

Thanks for the support.

albertolillo23, 2024-03-21:
Hi @Nicogene 
Today we tried it and found this error:

```
CMake Error at src/libYARP_eigen/src/CMakeLists. txt:25 (target_link_libraries):
	The link interface of target "VARP_eigen" contains:

		Eigen3:: Eigen

	but the target was not found. Possible reasons include:

          * ﻿﻿There is a typo in the target name.
          * ﻿﻿A find package call is missing for an IMPORTED target.
          * ﻿﻿An ALIAS target is missing.

CMake Error at src/libYARP_math/src/CMakeLists. txt:86 (target_link_libraries):
	Target "VARP_math" links™ to:

		Eigen3: : Eigen

        but the target was not found. Possible reasons include:

            * ﻿﻿There is a typo in the target name.
            * ﻿﻿A find package call is missing for an IMPORTED target.
            * ﻿﻿An ALIAS target is missing.

```
How can we solve this?

Thanks.

Nicogene, 2024-03-22:
Hi @albertolillo23,

To install the required dependencies, you have to follow these instructions:
- https://github.com/robotology/robotology-superbuild?tab=readme-ov-file#linux-from-source-with-dependencies-provided-by-apt

Here is instead the complete guide for installing from sources:
- https://github.com/robotology/robotology-superbuild?tab=readme-ov-file#source-installation

Nicogene, 2024-04-09:
Hi @albertolillo23,

> it can no longer find the other parts,

This is fine, you don't have to worry about it, the superbuild will clone and build the missing repositories.

> So as to start from a clean situation and download and build everything, upgrading the pc to a newer version of Ubuntu, now is 16.04.

We do not support Ubuntu 16.04, so yes I suggest you upgrade it to at least Ubuntu 20.04

Ccing @AntonioConsilvio for the upgrade procedure of the pc104

pattacini, 2024-04-10:
Hi @albertolillo23 

We had an internal meeting yesterday with our coordinator @maggia80 about the update of `iCubManchester01` and, given that it turned out to be quite old software-wise, we think it may be worth running a thorough maintenance.

In our experience, in fact, updating/upgrading the SW of a robot equipped with old HW is a painful procedure if carried out remotely. Also, while doing so, we often find that some HW intervention is required. 

Therefore, we suggest that you get in contact with our manager Andrea Derito (✉️ andrea.derito[at]iit.it) to ask for a quotation.

cc @andreaderito @Fabrizio69 @Nicogene

closed 2024-04-11 by pattacini