robotology/robots-configuration #643: [ergoCubSN00*] Missing elbow and wrist joints from axesNames used in WBD

traversaro, 2024-04-17:
See https://github.com/robotology/robots-configuration/blob/c6a298257827bb08de3c6b734ea7144399531df8/ergoCubSN001/estimators/wholebodydynamics.xml#L6 and similar list. 

I guess the following joints are missing:
* `l_elbow`/`r_elbow`
* `l_wrist_roll`/`r_wrist_roll`
* `l_wrist_pitch`/`r_wrist_yaw`
* `l_wrist_yaw`/`r_wrist_yaw`

wrist joints are probably just a nice to have, but the absence of the elbow joint may indeed affect the force and torque estimation.

cc @isorrentino @GiulioRomualdi @LoreMoretti @Nicogene @martinaxgloria

pattacini, 2024-04-18:
Is anyone eager to propose a PR, @isorrentino @GiulioRomualdi?

GiulioRomualdi, 2024-04-18:
I can do it

pattacini, 2024-06-18:
#656 added the missing info for ergoCubSN002.
I think ergoCubSN000 and ergoCubSN001 still need the patch.
cc @GiulioRomualdi

GiulioRomualdi, 2024-07-05:
ergoCubSN001 patched in https://github.com/icub-tech-iit/robots-configuration/commit/78c7b109712c108154d04beab2581bcc6b3a91d5

pattacini, 2024-07-05:
#664 added the patch for ergoCubSN001.
Closing for now.

closed 2024-07-05 by pattacini