robotology/robots-configuration #663: Enable tags not working for ros2

SimoneMic, 2024-07-02:
Hello,
I have tried the command for launching the YRI with ros2 enabled, as listed in this issue #641 https://github.com/robotology/robots-configuration/issues/641 

However the format is not recognised:
```
yarprobotinterface --config ergocub.xml --enable_tags (enable_ros2) 
-bash: syntax error near unexpected token `('
```
I have also tried removing the brackets, the robot starts, but Ros is not enabled.
Also by removing the --config part. Same result.

traversaro, 2024-07-02:
Can you try: 

~~~
yarprobotinterface --config ergocub.xml --enable_tags "(enable_ros2)"
~~~

Not sure why I did not experienced this problem, perhaps I tested on Windows.

pattacini, 2024-07-03:
Any feedback @SimoneMic?

SimoneMic, 2024-07-03:
I haven't used the robot today, when I'll have my slot I will test it

GiulioRomualdi, 2024-07-05:
```
yarprobotinterface --config ergocub.xml --disable_tags \(disable_right_arm\) 
```
works

pattacini, 2024-07-05:
Uhm, the format with the escape char `\` doesn't seem very clear in the first place, unless it's reported in the documentation. If not, it should be pointed out in YARP, somehow cc @randaz81. 

Out of curiosity, does it mean that "(disable_right_arm)" doesn't work?

SimoneMic, 2024-07-05:
With the "()" it works

randaz81, 2024-07-05:
I never heard about the use of \, I think it works by chance (undocumented and to be avoided.)
Instead if i remeber correctly you should always use " to encapsulate a  ( ) block, independently if it contains multiple elements or just one, like in your case.

pattacini, 2024-07-05:
Yeah, indeed, the standard ought to be `"(...)"`.
I remember now that I used it in the past for the `find-superquadric` module:
- https://github.com/robotology/find-superquadric?tab=readme-ov-file#example

I think we can close it.
Feel free to open it up again if you deem it necessary.

closed 2024-07-05 by pattacini

traversaro, 2024-07-08:
> The documentation should be changed, I think

Which documentation?